AsteroidOS/meta-smartwatch #12: Bluetooth not enable

ShuP1, 2022-03-06:
Is bluetooth support possible and/or plained ?
Every thing else works fine just date reset on reboot.

sergiosk39, 2022-05-05:
@anyc Thank you for such a pretty OS for my ZenWatch 3. Please enable bluetooth if it is possible! It'll give a whole new life to my zenwatch...)

MagneFire, 2022-07-28:
This issue is worth an update since I also did some digging into this specific issue a while back.

## Background

I think some background is needed since `swift` (Asus Zenwatch 3) is in the minority of not functioning Bluetooth.
Generally a dedicated Bluetooth chip is connected to the main processor by a serial interface. In our case most supported watches use a Qualcomm Snapdragon 400 or Wear 2100/3100 SoC. Most (maybe all?) of them use a Broadcom Bluetooth chip.

The biggest difference with `swift` compared to other watches is the serial interface. Most of them use a dedicated UART peripheral and kernel driver to communicate with the Bluetooth chip. As this is a generic solution, a separate kernel driver (`bluesleep` or more recently `nitrous`) is used to setup the interrupts so that the Bluetooth chip can wake the SoC when it sleeps. A userspace program named `brcm-patchram-plus`[^1] is responsible for flashing a firmware to the Bluetooth chip and enabling the low power mode.

However, `swift` uses something known as `Qualcomm SMD`[^5]. This means  that something embedded in the SoC handles the communication with the Bluetooth chip. A kernel driver still handles the communication but in addition also handles the flashing of the firmware to the Bluetooth chip and interrupts are also handled by the same driver. I'm not sure if it handles the setup of the low power modes, this may be part of an Android binary. After some reverse engineering of those binaries I did not find anything useful.

## Bluebinder

We've had newer watches that support Bluetooth that appear to be close to identical when it comes to the hardware configuration.
With these newer watches we use `bluebinder`[^2] to directly talk to the Android libraries responsible for setting up Bluetooth. This will, however, only work on Android 8 bases and above. `swift` unfortunately never got an update to this version.

I also own a `ray` (Skagen Falster 2) that uses the same SoC, it also uses Qualcomm SMD to setup the Bluetooth chip and it's required low power mode. Due to the lack of support for `bluebinder` on `swift` I decided to port `swift` from an Android 6 base to the Android 7 base. This will reduce the chance of missing symbols in binaries when you try to execute Android binaries from another watch. Realistically, it meant that I could execute the required Android binary[^3] that would setup the Bluetooth device and allow for communication with `bluebinder`. But it also required my to backport the `binder` kernel driver[^4] from `ray` to `swift`. All this worked, but it lead to the original issue where you could pair your phone with the watch but it would disconnect after that.
This make me suspicious of the Bluetooth firmware files used. But I'm not entirely sure how to debug firmware related issues. I may try to switch to the `narwhal` binary as it appears to be a little bit different.

[^1]: https://github.com/AsteroidOS/brcm-patchram-plus/blob/master/src/main.c
[^5]: https://osmocom.org/projects/quectel-modems/wiki/Qualcomm_Linux_SMD
[^2]: https://github.com/mer-hybris/bluebinder
[^3]: https://github.com/AsteroidOS/meta-smartwatch/blob/master/meta-ray/recipes-android/android-init/android-init/init.rc#L49
[^4]: https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-ray-3.18-oreo-wear-dr/drivers/staging/android/binder.c

loopaware, 2022-08-30:
Thank you for your hard work, please keep it up! :) I have a handful of zenwatch 3, all running Asteroid OS. Dreaming of a future where the bluetooth works on them.