AsteroidOS/meta-smartwatch #35: Watch never enters deepsleep.

MagneFire, 2022-03-06:


Doomsdayrs, 2022-03-27:
```journalctl -f```
Attempting to run `systemctl suspend`
[2022-03-27-suspend-failure.txt](https://github.com/AsteroidOS/meta-smartwatch/files/8358188/2022-03-27-suspend-failure.txt)

Doomsdayrs, 2022-03-27:
Random errors that gets pumped out on cycle
```
Mar 27 14:02:31 tetra kernel: bcmpmu_rpc_work, pa_temp 384
```
```
Mar 27 14:02:31 tetra kernel: ADC out of range, raw = 0
```
```
Mar 27 14:02:35 tetra kernel: bcmpmu_read_time: err=0 time=122.2.27.14.2.53
```

Doomsdayrs, 2022-03-27:
```journalctl --system -f```
[2022-03-27-journalctl.log](https://github.com/AsteroidOS/meta-smartwatch/files/8358199/2022-03-27-journalctl.log)

The MCE is keeping the CPU alive? Why?
```
Mar 27 14:07:00 tetra mce[314]: modules/cpu-keepalive.c: cka_state_set(): keepalive started
```

Pepa77, 2022-07-17:
hi i don't know how it is possible but last night 15/07/2022 tetra sleeps like not to mention after one days i have 33% battery now Asteroid OS has chance again on this device.

mirh, 2022-07-17:
There's literally a PR linked to explain how it was possible.

Pepa77, 2022-07-17:
Yeah sorry I missed it.

Doomsdayrs, 2022-07-17:
Now to have tetra re-added to the website

eLtMosen, 2022-07-18:
> Now to have tetra re-added to the website

Since there are still problems with the gpu, @MagneFire proposed to add watches with remaining problems like minnow and tetra to a new "Experimental" section. That way, users would not falsly expect a daily driver ready support level and possibly make a purchase based on those watchs appearing in the general install pages. And developers have an easier time finding it still.