AsteroidOS/meta-smartwatch #55: Make sensors work

atx, 2022-03-06:
The connected sensors are described in [sensor_def_lenok.conf](https://p.atx.name/yvhekm2mn7). However, they are not connected directly to the ARM core, but instead to the Hexagon DSP coprocessor.

The kernel loads the (full of debug printf strings!) firmware to the coprocessor from `/system/vendor/firmware/adsb.*`, and then setups several /dev files which are presumably used by the original Android code to interface with the coprocessor using [FastRPC](https://developer.qualcomm.com/blog/how-use-fastrpc-offload-cpu-qualcomm-hexagon-dsp).

The /dev/adsprpc-smd ioctls are described in [the kernel documentation](https://android.googlesource.com/kernel/msm/+/android-msm-flo-3.4-kitkat-mr1/Documentation/arm/msm/adsprpc-drv.txt). What is missing is description of the RPC methods provided by the LG code.

These could likely be reverse engineered by statically analyzing the [DSP firmware](https://github.com/programa-stic/hexag00n) or by stracing ioctl calls in Android.

docgalaxyblock, 2022-09-27:
Any news on this issues state?
My watches sensors are working

Are @GeckoEidechse problems resolved (somewhere else)?

lenok build 19.9.'22
---
<details><summary><code>test_sensors</code></summary><pre><code>
Hardware module ID: sensors
Hardware module Name: QTI Sensors Module
Hardware module Author: Qualcomm Technologies, Inc.
Hardware module API version: 0x1
Hardware HAL API version: 0x100
Poll device version: 0x1030001
API VERSION 0.1 (legacy): 0x1
API VERSION 0.1: 0x65537
API VERSION 1.0: 0x16777217
API VERSION 1.1: 0x16842753
API VERSION 1.2: 0x16908289
API VERSION 1.3: 0x16973825
API VERSION 1.4: 0x17039361
Got 33 sensors

=== Sensor 0 ==
Name: MPU6515 Accelerometer
Vendor: InvenSense
Version: 0x1
Handle: 0x0
Type: 1
maxRange: 19.613296
resolution: 0.595
power: 0.400000 mA
minDelay: 5000

=== Sensor 1 ==
Name: AK8963 Magnetometer
Vendor: AKM
Version: 0x1
Handle: 0xa
Type: 2
maxRange: 4911.999511
resolution: 0.149536
power: 5.0 mA
minDelay: 16666

=== Sensor 2 ==
Name: AK8963 Magnetometer Uncalibrated
Vendor: AKM
Version: 0x1
Handle: 0xb
Type: 14
maxRange: 4911.999511
resolution: 0.149536
power: 5.0 mA
minDelay: 16666

=== Sensor 3 ==
Name: MPU6515 Gyroscope
Vendor: InvenSense
Version: 0x1
Handle: 0x2
Type: 4
maxRange: 34.906585
resolution: 0.1068
power: 3.200000 mA
minDelay: 5000

=== Sensor 4 ==
Name: MPU6515 Gyroscope Uncalibrated
Vendor: InvenSense
[... 432 lines cut ...]
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  12  in  96  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  12  in  96  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  13  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  13  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  14  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  14  in  101  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  15  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  15  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  16  in  95  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  16  in  95  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  17  in  101  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  17  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  18  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  18  in  101  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  19  in  102  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  19  in  103  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  20  in  96  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  20  in  95  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  21  in  103  ms
QDEBUG : ClientApiTest::testDownsamplingDisabled() dataAvailable()  21  in  104  ms
PASS   : ClientApiTest::testDownsamplingDisabled()
PASS   : ClientApiTest::cleanupTestCase()
Totals: 16 passed, 5 failed, 0 skipped, 0 blacklisted, 18699ms
********* Finished testing of ClientApiTest *********
</code></pre></details>

eLtMosen, 2022-09-27:
Mine have been since long time. Only ever flashed nightlies and never encountered not working sensors on lenok.

GeckoEidechse, 2022-09-29:
> Any news on this issues state? My watches sensors are working
> 
> Are @GeckoEidechse problems resolved (somewhere else)?

Nope, though I haven't attempted flashing back WearOS yet to update my lenok's firmware.

GeckoEidechse, 2023-02-22:
Solved my particular issue in #159

eLtMosen, 2023-02-22:
Guess its time to close both issues then?

MagneFire, 2023-02-22:
No, I think we should update our website. Only after that has been done close the issues.

eLtMosen, 2023-02-22:
> No, I think we should update our website. Only after that has been done close the issues.

Agreed ;)

closed 2023-02-22 by MagneFire