AsteroidOS/meta-smartwatch #56: Can't connect mooneye with usb

wannaphong, 2022-03-06:


MagneFire, 2022-04-17:
We probably shouldn't close this just yet as the 'solution' hasn't been tested.

wannaphong, 2022-04-18:
I will wait lasted build to test mooneye after merged patch. https://release.asteroidos.org/nightlies/mooneye/

wannaphong, 2022-04-23:
@MagneFire It doesn't working.

fluffysfriends, 2023-04-16:
I tried to access this on Ubuntu Desktop and i mange to connect using SSH

eLtMosen, 2023-04-16:
> I tried to access this on Ubuntu Desktop and i mange to connect using SSH

Can confirm.
But also note that ADB still does not.
@MagneFire requested some logs an analyzed them, coming to the conclusion that there are several android services constantly crashing, leading to the UI being software rendered even afaiu.
The mooneye port has to get seriously polished :D