AsteroidOS/meta-smartwatch #57: Bt works , but didn't connect to my phone

ItsVixano, 2022-03-06:
hi , i'm using a mooneye , and after installed it , i tried to connect to my phone but , he make a loop of connect/disconnect 
i love this project and i want to use it every day , but i cannot do this
it see be a phone issue "Whyred with EvoX" but i don't know 

i'm using the last snapshot release

sorry for my grammar , i'm italian

fluffysfriends, 2023-04-15:
Any news regarding this issue?

fluffysfriends, 2023-04-16:
hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.65
device: hci0 snap_len: 1500 filter: 0xffffffff
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 512 reason 0x08
    Reason: Connection Timeout
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 512 reason 0x08
    Reason: Connection Timeout
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Parameters (0x08|0x0006) ncmd 1
    status 0x00
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 1
    status 0x00
I tried to connect two times using Gadgetbridge on Android 13 using a Sony Xperia X10 III

MagneFire, 2023-04-16:
> Any news regarding this issue?

Nope, last time I checked the proposed idea of using `bluebinder` won't work since `mooneye` doesn't use the `/dev/hwbinder` interface. This means there's work required on the kernel driver to figure out what's going wrong.