AsteroidOS/meta-smartwatch #58: No sound or vibration when timer ends

saladhax, 2022-03-06:
Hey, so this might be kind of a small stupid thing, but I noticed that when I set a timer on AsteroidOS on my mooneye watch, there's no sound or vibrations when the timer finishes. A little popup thing will appear saying the timer has finished, but that's it. I have the little vibrations toggle turned on in the quick settings panel (the thing that appears when you swipe down from the watch face) but it still doesn't work. Any ideas on what might be going on?

fluffysfriends, 2023-04-15:
I have tried the latestet beta and in that i have vibration when the alarm ends

eLtMosen, 2023-04-15:
@fluffysfriends could you clarify if that is indeed working on a Ticwatch E or S (mooneye)?
The issue only has mooneye as label and its not totally obvious that this issue is only mooneye specific.
Other watches do not have vibration issues currently.

fluffysfriends, 2023-04-15:
@eLtMosen I have tried the alarm app and i get vibration when the alarm is triggerd. So i saw this issue and made a replay. And the watch that i am using is a ticwatch s

eLtMosen, 2023-04-15:
> @eLtMosen I have tried the alarm app and i get vibration when the alarm is triggerd. So i saw this issue and made a replay. And the watch that i am using is a ticwatch s

Perfect, thank you! Going to double check on my Ticwatch E then and report back.

eLtMosen, 2023-04-15:
https://user-images.githubusercontent.com/15074193/232243901-2fb35e70-e0f8-4701-8828-c7cdf1762b55.mp4

Don't know if one can hear it in the video. But vibration works when timer and alarm presenter get triggered.
Thank you for pointing this out @fluffysfriends 
Much appreciated!

closed 2023-04-15 by eLtMosen