AsteroidOS/meta-smartwatch #68: [catfish] Time on the LCD is wrong in temporary installation and gets reset every reboot.

locksmxth5839, 2022-04-01:
Good {morning,afternoon,evening},
I just ('just' meaning Apr. 1, 2022) took a look at the install page of Asteroid OS and found that the TicWatch Pro is supported, so I did the temporary installation, and it booted and worked perfectly fine. However, with every reboot, the time changes back to the default time, which is around 11 something on Mar. 17, and when I change the time, the clock on the LCD doesn't seem to respond to the changes. I don't know what's exactly going on here...

Please let me know if you can fix this, if there's a patch, or if I have to do the real installation to fix this.

Thanks,
LockSmxth5839

frankenstein91, 2022-04-02:
Even with a permanent installation, the time is always reset on reboot. Good news, however, the time zone in timedatectl remains.

frankenstein91, 2022-04-03:
I just noticed something... when rebooting, the clock does not lose all the time... The year of the date remains. Only the information on the LCD display is changed, i.e. hour, minute, day and month. The year is not displayed on the LCD and is not overwritten.

locksmxth5839, 2022-04-07:
Yes I noticed that too... Well, I don't really care about the LCD, because I can override the LCD with Always-On Display, but that would drain a ton of battery.

LecrisUT, 2022-04-07:
Please wait for #65. I'll finish it up this weekend.

SEVA77, 2023-04-02:
I tried to find a reason but Im not sure about my search results.
Why `dsme.service` does not give access to RTC time for other services?
```
$ adb shell
/ # timedatectl
               Local time: Sun 2023-04-02 02:26:05 +03
           Universal time: Sat 2023-04-01 23:26:05 UTC
                 RTC time: n/a
                Time zone: Europe/Minsk (+03, +0300)
System clock synchronized: no
              NTP service: n/a
          RTC in local TZ: no
/ # hwclock --show
hwclock: can't open '/dev/rtc': Device or resource busy
/ # systemctl stop dsme.service
/ # hwclock --show
Thu Jan  1 14:02:24 1970  0.000000 seconds
/ # timedatectl
               Local time: Sun 2023-04-02 02:27:50 +03
           Universal time: Sat 2023-04-01 23:27:50 UTC
                 RTC time: Thu 1970-01-01 14:02:29
                Time zone: Europe/Minsk (+03, +0300)
System clock synchronized: no
              NTP service: n/a
          RTC in local TZ: no
```

PR #177 (2023-04-02): catfish: Install swclock-offset.
The time isn't saved to the RTC on `catfish` so we need to use `swclock-offset` to fix this.

This fixes #68.
closed 2023-04-02 by FlorentRevest