AsteroidOS/meta-smartwatch #69: [catfish] Heart Rate not working

frankenstein91, 2022-04-02:
Contrary to the documentation on the website, measuring the heart rate does not work.
The load bar in the corresponding app moves, but unfortunately no measurement is performed and not even the LED at the bottom of the watch is turned on.

MagneFire, 2022-04-03:
Can you clarify if you are using the 2018 (catfish) or 2020 (catfish_ext) variant?
Recently, we updated the sensor firmware to fix sensors (heart rate, tilt-to-wake, compass) on the 2020 version. It's possible that this lead to things breaking for the 2018 version.

frankenstein91, 2022-04-03:
I am not sure when I bought the watch. But the bootloader only says catfish.

frankenstein91, 2022-04-03:
After installing the file nanohub.full.catshark.19061201.bin from @MagneFire, the problem is solved.

(I think until the next update)

frankenstein91, 2022-04-03:
log of `journalctl --no-pager`
[log.log](https://github.com/AsteroidOS/meta-smartwatch/files/8405281/log.log)

maybe also usefull for https://github.com/AsteroidOS/meta-smartwatch/issues/71 and https://github.com/AsteroidOS/meta-smartwatch/issues/70

abhinavmarwaha, 2022-05-15:
> After installing the file nanohub.full.catshark.19061201.bin from @MagneFire, the problem is solved.
> 
> (I think until the next update)

How to do that exactly? I installed the one from website, in that it doesn't work.

frankenstein91, 2022-05-15:
> How to do that exactly?

copy & past via SCP and than a reboot of the watch

abhinavmarwaha, 2022-05-16:
> > How to do that exactly?
> 
> copy & past via SCP and than a reboot of the watch

where can I find this file?

sevmonster, 2023-02-28:
New user, I do not have this issue on the latest nightly. Heart rate sensor works fine.

eLtMosen, 2023-03-01:
Agreed, double checked on my catfish, works fine. Closing, feel free to reopen or make a new issue when the problem should arise again.

closed 2023-03-01 by eLtMosen