AsteroidOS/meta-smartwatch #70: [catfish] Battery charge hangs at xx%

frankenstein91, 2022-04-02:
I had to find out that the battery charge indicator doesn't work quite as well yet.
At the first attempts the charge level hung at 12% after a reboot now at 50%.

Removing the loading dock is detected properly.

```
cat /sys/class/power_supply/battery/capacity
50
cat /sys/class/power_supply/battery/health
Good
cat /sys/class/power_supply/battery/status
Full
```

frankenstein91, 2022-04-03:
After installing the file nanohub.full.catshark.19061201.bin from @MagneFire, the problem is solved.

battery status is now okay in `/sys/class/power_supply/nanohub_fuelgauge-0/` but not in `/sys/class/power_supply/battery`/capacity

In GUI it looks fine

jlawsonusmc, 2022-05-06:
> After installing the file nanohub.full.catshark.19061201.bin from @MagneFire, the problem is solved.
> 
> battery status is now okay in `/sys/class/power_supply/nanohub_fuelgauge-0/` but not in `/sys/class/power_supply/battery`/capacity
> 
> In GUI it looks fine

I'm a complete novice, but am having the same issue. Can you point me in the right direction so I can replicate your fix?

beroset, 2024-05-14:
This https://github.com/AsteroidOS/meta-smartwatch/commit/61abaa7bf60a33804b175363574a4e6d9c4c42ab commit fixed this problem in 2022.

closed 2024-05-14 by beroset