AsteroidOS/meta-smartwatch #77: Support for oppo watch international version

wannaphong, 2022-04-19:
I found it can unlock oem with latest update and I found source code kernel.
[image]
[image]

source code kernel https://android.googlesource.com/kernel/msm/+/refs/heads/android-msm-beluga-4.9-pie-wear-mr2

iot-resister, 2022-04-26:
Can I update the docs for this?

wannaphong, 2022-04-27:
The real installation is working!

nelikp222, 2022-07-17:
Thank you for this work
both of you @wannaphong and @MagneFire 
is it safe to install now?
And any stock rom i can download incase of anything?

wannaphong, 2022-07-17:
> Thank you for this work both of you @wannaphong and @MagneFire is it safe to install now? And any stock rom i can download incase of anything?

You can download the stock rom of oppo watch from Google. I can't upload to you becasue legal requirements. It's `.ofp` and can use https://github.com/bkerler/oppo_decrypt for find the stock rom.

wannaphong, 2023-01-05:
I closed this issues because all list support are done by @MagneFire. Thank you!

closed 2023-01-05 by wannaphong