AsteroidOS/meta-smartwatch #82: Heart rate sensor no longer works

MagneFire, 2022-05-10:
It seems that https://github.com/AsteroidOS/meta-sturgeon-hybris/issues/6 has returned...

The hear rate sensor appears to light up and report a heart rate until you restart the heart rate app. At that point `sensorfw` reports an issue where `setInterval` was `not ok`.
My previous findings were that this function is essential for `sturgeon`.
I'll check if downgrading `sensorfw` to an older version fixes this issue again then we can look into creating a fix based on the current version of `sensorfw`.

This issue was reported by @jrtberlin too.

Pepa77, 2022-06-07:
hello I can confirm that the sensor works only once after restarting the watch, the second time I will not run on the night release of firmware 1.1 for sturgeon 3.6.2022

eLtMosen, 2022-06-07:
Confirmed on my sturgeon. It works one time (does not matter how long after reboot, the watch  ran the whole day before i tried now). Closing the app and trying again does not light up the sensor.

Pepa77, 2022-06-11:
hello it would fair on the site settings Asteroid OS for sturgeon heart rate put on red that it broken.

MagneFire, 2022-06-11:
> hello it would fair on the site settings Asteroid OS for sturgeon heart rate put on red that it broken.

Agreed, I shall update the website until we have a fix.

MagneFire, 2022-08-05:
I've done some testing and I think the heart rate sensor stopped functioning reliably after this commit: https://github.com/AsteroidOS/meta-asteroid/commit/e3a4c38b5ec3c49d5330546ce0ec232535e0b4d1.

The current plan is to downgrade `sensorfw` prior to this change and test if this makes the heart rate sensor function correctly on current nightlies.

closed 2022-08-24 by FlorentRevest