AsteroidOS/meta-smartwatch #84: [catfish] Sensors stop working on latest builds

LecrisUT, 2022-05-11:
Recent builds appear to brick `nanohub` when performing an update (`/vendor/bin/nanoapp_cmd download`). The only difference in the builds on my side were according to #83. Still there is no guarantee that others who are doing a fresh boot from the nightlies which would trigger the nanohub update won't be affected. If others have done so recently and had (no) problems, it would be helpful to know.

The source of this issue is a mystery currently as the recent commits do not have an evident impact there.

### For now be careful when installing on `catfish`.

dodoradio, 2022-05-16:
After further investigation (and another bricked catfish) it seems that the issue is caused by nanohub being flashed repeatedly by dual booting (upgraded by WearOS and downgraded by AsteroidOS) or by some downgrade protection. Either way, this seems to be a hardware problem, or at least a firmware problem that is separate from the kernel. 
A quick remedy for this is to remove the [nanohub updating service from init.rc](https://github.com/AsteroidOS/meta-smartwatch/blob/master/meta-catfish/recipes-android/android-init/android-init/init.rc#L59) while we figure out what's breaking. This has no side effects, catshark seems to still boot fine without this service and nanohub functions fully (dual display turns on, sensors work).

closed 2022-05-16 by MagneFire

LecrisUT, 2022-05-16:
Let's keep this open until I can confirm it can run with different binaries.