AsteroidOS/meta-smartwatch #91: [sawfish] Vibration not working

domkrm, 2022-06-03:
First of all, thanks for this great project and maintaining the Huawei Watch 2.

Today I installed AsteroidOS on my watch. All is working fine, except the vibration.

If I get a notification or toggle the vibration in the menu, the watch doesn't vibrate. Under Wear OS the vibration was working fine.

Is vibration supported in AsteroidOS by sawfish at all? If so, what can I do to analyze and debug the problem?

MagneFire, 2022-06-04:
Thanks for reporting this issue!
I've had a look into this and the cause has been found.

Once https://github.com/AsteroidOS/meta-asteroid/pull/102 is merged and a new nightly is released, this issue should be solved.

docgalaxyblock, 2022-09-28:
It sounds like you issue was resolved? if so could you close it?

To your question:
afaik the buildsystem does not trigger a complete rebuild of all watches if only specific ones get a change

domkrm, 2022-10-14:
Yes, issue is resolved. Closing...

closed 2022-10-14 by domkrm