AsteroidOS/meta-smartwatch #92: asteroid-image-wren.ext4 error max download size of 63963136 bytes

seisonimi, 2022-06-05:
As in issure 31 stated the error does occur on the newest nightly for the Asus Zenwatch 2.
I can only enter CSC Fastboot Mode right now but couldn't get the ext2simg to work.

The error I get there is 
C:\Android>ext2simg asteroid-image-wren.ext4 asteroid.simg
error: write_data_file: mmap64: Invalid argument

Are there new releases with the simg version planned or is there another way to convert them manually?

Pepa77, 2022-06-07:
Hi I had the same problem on the night edition for sturgeon, but I needed to go do something else I let it be and when I got back to the computer after about half an hour it was installed 😁

seisonimi, 2022-06-16:
I got it to work:
The problem only occurred when I used Windows ADB Tools.
This time I used Linux (Ubuntu) which did not give me an error for max download size and let me flash userdata as well as boot without any problems whatsoever.

MagneFire, 2022-07-29:
Great to see you fixed this. We've seen these strange issues before as well, but always only when people use Windows based systems.

For future reference, updating the ADB tools may also solve this issue.

I'll close this issue as it's more related to something Windows specific and not AsteroidOS. But feel free to re-open it if this remains an issue :smile:

closed 2022-07-29 by MagneFire