AsteroidOS/meta-smartwatch #96: Dory nightly from 17th June issues

megamaced, 2022-06-25:


eLtMosen, 2022-06-26:
Indeed, we got some regressions.
When trying to open either wallpaper, watchface or starter pages from settings, the settings page closes.
`Journalctl -f`reports:
`invoker[917]: error: Can't send signal 11 to application [899]: No such process`

Also, the `ssh` connection aka Developer Mode does not work. Mode is enabled on the watch and claims "SDK mode activated" when connecting via USB. But no network is announced on the connected computer.

@megamaced to close Diamonds, just swipe from top. That does work for me.

Agenda does not start at all for me, `Journalctl -f` shows
```
Jun 26 17:49:57 dory asteroid-launcher[956]: ../../git/src/sqlitestorage.cpp: 193 - database "/home/ceres/.local/share/system/privileged/Calendar/mkcal/db" opened
Jun 26 17:49:57 dory mapplauncherd[465]: Boosted process (pid=956) was terminated due to signal 11
Jun 26 17:49:57 dory invoker[971]: error: Can't send signal 11 to application [956]: No such process
Jun 26 17:49:58 dory booster-qtcomponents-qt5[972]: qt.qpa.wayland: qtvirtualkeyboard currently is not supported at client-side, use QT_IM_MODULE=qtvirtualkeyboard at compositor-side.
Jun 26 17:50:00 dory mce[291]: modules/alarm.c: alarm_queue_status_dbus_cb(): Received alarm queue status signal from name=:1.15 owner=:1.15 pid=410 uid=1000 gid=1000 priv=0 cmd=/usr/bin/timed-qt5 --systemd
```

All other apps but agenda work as expected on my dory with 2022-06-24 Nightly.

jrtberlin, 2022-07-23:
I can confirm the issues. I have just flashed the most recent nightly and I'm experiencing the same issues.

beroset, 2022-10-04:
I have traced the problem to the `DropShadow` portion of `WallpaperPage.qml`.  If I comment that out, wallpaper selection works again.  I don't yet know what is wrong within `DropShadow`.

beroset, 2022-10-09:
Made some progress, but no fix yet.  Maybe this info will help someone else figure it out.

First, I created a debug build of the `tst_qtgraphicaleffects` executable that is bundled with the `qtgraphicaleffects` package, but not normally installed.  I did this on my development machine by fist executing `bitbake -c devshell qtgraphicaleffects` and then:
```
mkdir build
cd build
qmake CONFIG+=force_debug_info  ../qtgraphicaleffects.pro
make -j sub-tests
scp tests/auto/tst_qtgraphicaleffects ceres@192.168.2.15:.
```
I also installed `gdb` on the watch using `opkg` and enabled core dumps using `ulimit -S -c unlimited`.

Then I logged into the watch as the `ceres` user and ran the program:
```
EGL_PLATFORM=wayland QT_QPA_PLATFORM=wayland ./tst_qtgraphicaleffects
```
It runs for a bit and then encounters a segfault and dumps backtraces of multiple threads. Installing debug versions of `libhybris` helps a bit.  With some experimentation, I discovered that the tests that fail are:

- dropShadow
- gaussianBlur
- glow
- maskedBlur

All of these ultimately derive from GaussianBlur (Glow derives from DropShadow which includes GaussianBlur).

beroset, 2022-10-10:
Based on a comment from @MagneFire I reverted to the commit immediately before this one:  https://github.com/meta-qt5/meta-qt5/commit/74c27fa59b329f1210d39e8cd2904acf30d10838#diff-4d0e700d3ee6cb517d3cd2506ed082227ea798df8edcb34e776e58b07b1881b2

Rebuilding the image for `dory` with only this one change restores correct operation and fixes the bug.  So now we need to figure out what is is within this that is causing a problem, and why only `dory`.

MagneFire, 2023-01-22:
This has been solved as of: https://github.com/AsteroidOS/meta-smartwatch/pull/148

closed 2023-01-22 by MagneFire