AsteroidOS/meta-smartwatch #97: LG G Watch R stuck in boot loop

jackkaback, 2022-06-25:
My LG G Watch R is stuck in a boot loop while connected to the dock. It boots, I get the LG logo, and then I get the AsteroidOS logo before repeating.

eLtMosen, 2022-06-25:
Is this a fresh flash? Then i would just try to flash again. Connection problems sometimes cause faulty flash processes.
Or did you run this lenok (LG Watch R) since longer time already with AOS and the boot loop appeared just now?
Did you possibly update via OPKG?
Thanks for reporting and helping to narrow the issue in.

jackkaback, 2022-06-25:
Yes this is a fresh install, though when I am on the dock it's stuck in the boot loop. out of the dock it appears to only go to the LG screen, so I'm guessing it's too low on battery now. Even after connecting it to the main it only boot loops while getting hot.

jackkaback, 2022-06-25:
Well, I let it stay attached to main for a while and it was stuck, so I did try doing a re-flashing, now however if I disconnect it from the dock it immediately powers down, even though it has 49% battery. Before putting on Asteroid it was able to stay on without being docked.

jackkaback, 2022-06-25:
Scratch that, now it wants to work. Thanks for the assistance.

closed 2022-06-25 by jackkaback