AsteroidOS/meta-smartwatch #104: Low UI performance

MagneFire, 2022-07-31:
`minnow` is suffering from an issue where the framerate is very low (<10).
Part of the reason is the use of `QSG_NO_DEPTH_BUFFER=1` (https://github.com/AsteroidOS/meta-smartwatch/blob/5b73918c11659981f2ec2be5cd18acc3955ec811/meta-minnow/recipes-asteroid/asteroid-launcher/asteroid-launcher/default.conf#L6). This is needed as otherwise only the FlatMesh/background is shown.

Without setting said variable you can still launch apps, but they are still a bit slow though not as slow when `QSG_NO_DEPTH_BUFFER` is set.

When `QSG_NO_DEPTH_BUFFER` is NOT set the homescreen looks like this:

[image]

## Current findings

### smoothstep() shader issue.
While porting I had the `MACHINE_DISPLAY_ROUND` enabled. This resulted in a vague rendering of apps:
[image]

Round watches have a shader applied so that all windows appear round (supposedly improving performance as well).
(https://github.com/AsteroidOS/asteroid-launcher/blob/2e0048ac90559a7eb66a80b8e38da6fb816383c9/qml/compositor/WindowWrapperBase.qml#L59)

After some debugging it turns out that the use of `smoothstep` causes for the strange rendering issue to appear. Removing the use of `smoothstep()` (https://github.com/AsteroidOS/asteroid-launcher/blob/2e0048ac90559a7eb66a80b8e38da6fb816383c9/qml/compositor/CircleMaskShader.qml#L62) solves this issue but makes the window look not as smooth when you swipe an app to close it.

### Z-order is not respected

The issue where only the wallpaper is shown has something to do with the ordering of the elements. As removing to the wallpaper (https://github.com/AsteroidOS/asteroid-launcher/blob/2e0048ac90559a7eb66a80b8e38da6fb816383c9/qml/MainScreen.qml#L378) seems to solve this specific issue.

The code uses the `z` property to place it behind the `grid`, but this does not appear to work. Manually moving the wallpaper to above the `PanelsGrid` also does not solve this. This issue seems to have something to do with the opacities of elements as placing a simple `Rectangle` on screen is also always shown on top of the `grid` regardless of `z` order or placement in code. The only time it's rendered below the grid is when the `opacity` of the rectangle was reduced. I have no idea why this is happening https://stackoverflow.com/questions/60498229/qt-qml-opengl-z-ordering-doesnt-work-with-semi-transparent-items-with-opengl-u shows a similar behaviour of what is happening here. This needs further research.

That being said even if we solve above issues, the UI performance will still be slow. It may have something to do with the older hwcomposer (v1.0 instead of the more common v1.1).

## Other notes

The framebuffer reports a framerate of 45 fps as reported by WearOS.
The `qt5-qpa-hwcomposer-plugin` tries to get the framerate from the hwcomposer instead of the framebuffer (https://github.com/mer-hybris/qt5-qpa-hwcomposer-plugin/blob/master/hwcomposer/hwcomposer_backend_v10.cpp#L275). Which is a deprecated method in the v1.0 hwcomposer. The `minnow` hwcomposer doesn't support this. This results in it defaulting to a framerate of 60 fps. Which is incorrect. Manually changing this to 45 fps doesn't change anything.

It's worth mentioning that the default used is incorrect (https://github.com/mer-hybris/qt5-qpa-hwcomposer-plugin/blob/master/hwcomposer/hwcomposer_backend_v10.cpp#L278) as it sets the VSync period to 60 instead of 16.7 milliseconds. This does not have an impact on the performance though.

FlorentRevest, 2022-08-01:
Great findings here already :) I think it's alright to introduce watch-specific hacks in the minnow layer if this leads to a usable system. We can also keep minnow in the same category of support as tetra

MagneFire, 2023-03-07:
> Is there any news about (minnow) watch? Is it still in progress or it's development stopped already?

Nothing major. Every once in a while I debug this issue in hopes of a good solution. But nothing as of yet.