AsteroidOS/meta-smartwatch #106: [lennok] Bootloop if the battery is empty

ManuelBoe14, 2022-09-08:
**Summary:**
Whenever the battery of the watch is empty and I try to recharge it, I notice that asteroidos can no longer be started. instead, the bootloader is stuck in a kind of bootloop.

**Possible causes:**
The power consumption of the asteroidos is probably higher than the clock can provide without a battery.

**Steps to reproduce:**
1. Discharge the battery of the watch
2. charge the watch
3. the boot loop occurs.
4. go to the boot menu to charge the battery a little.
5. the watch can be started normally again

dodoradio, 2022-09-08:
I have seen a similar issue on narwhal. Not sure if this is entirely relevant, but the behaviour is as follows:
- The watch boots to asteroid
- A low battery warning is shown
- A 'insufficient charging current' message is shown
- A 'critical battery level' warning is shown
- the watch shuts down
- the watch reboots

ManuelBoe14, 2022-09-10:
Sorry for the late reply. i had to wait for my the battery to drain before i could do the tests.

**When it bootloops, does it show the AsteroidOS logo for a short period?**
Yes, the asteroids logo and the watchface are visible for a few seconds.

**Does it boot successfully into AsteroidOS now?**
Yes, after following the steps above, I was able to boot successfully into AsteroidOS.

ManuelBoe14, 2022-09-10:
> I have seen a similar issue on narwhal. Not sure if this is entirely relevant, but the behaviour is as follows:
>
>     * The watch boots to asteroid
>
>     * A low battery warning is shown
>
>     * A [...]

The issue on narwhal seems to be identical.