AsteroidOS/meta-smartwatch #114: Same Bluetooth MAC used for multiple watches

eLtMosen, 2022-10-06:
At least Lenok and Bass show the same BT-MAC when observed.
Its `43:43:A0:12:1F:AC`.
This makes it hard to pair those watches in succession.

@MagneFire noted on matrix that in some ports BT MAC
> are generally restored from the persist or firnware partition.
> Basically all watches that have support for Bluetooth and don't use the bluebinder might have this issue. sturgeon is amongst them

eLtMosen, 2023-01-12:
Sparrow and wren also have this issue

MagneFire, 2023-01-12:
I think the following watches are affected by this:
- `anthias`
- `bass`
- `dory`
- `lenok`
- `sawfish`
- `smelt`
- `sparrow`
- `sturgeon`
- `wren`

Basically all watches that use `brcm-patchram-plus` (https://github.com/AsteroidOS/brcm-patchram-plus) to flash the Bluetooth firmware. These watches also implement a different kernel driver: https://raw.githubusercontent.com/AsteroidOS/meta-smartwatch/master/meta-smelt/recipes-kernel/linux/linux-smelt/0005-Backport-mainline-4.1-Bluetooth-subsystem.patch (notice that it patches the Bluetooth drivers in addition to the Bluetooth network stack)

closed 2023-02-07 by FlorentRevest