AsteroidOS/meta-smartwatch #123: [Tetra] Cant mount mmcblkop32

Doomsdayrs, 2022-10-17:
```
Oct 16 20:55:16 tetra kernel: EXT4-fs (mmcblk0p32): couldn't mount as ext3 due to feature incompatibilities
Oct 16 20:55:16 tetra kernel: EXT4-fs (mmcblk0p32): couldn't mount as ext2 due to feature incompatibilities
```

MagneFire, 2022-11-06:
`mmcblkop32` is the userdata partition. Without this you won't have a functioning AsteroidOS watch. The boot script just tries to mount the userdata partition using an `auto` type: https://github.com/AsteroidOS/meta-asteroid/blob/971762769951f621cfdc5682cb0b2af8232cbab3/recipes-core/initrdscripts/initramfs-scripts-android/init.sh#L83

dxkf177, 2024-06-03:
> ```
> Oct 16 20:55:16 tetra kernel: EXT4-fs (mmcblk0p32): couldn't mount as ext3 due to feature incompatibilities
> Oct 16 20:55:16 tetra kernel: EXT4-fs (mmcblk0p32): couldn't mount as ext2 due to feature incompatibilities
> ```