AsteroidOS/meta-smartwatch #129: [Lenok] After an update Watch is stuck on the logo screen

jackkaback, 2022-10-21:
I ran an update to my watch and afterwards it started acting buggy so I gave it the old power cycle. Afterwards it's stuck on the splash screen for ASteroidos, I'm able to SSH into the watch so it is booted, but it's just stuck on this screen

jackkaback, 2022-10-21:
Running "top" in the watch reveals that process ceres is using some 95% of the CPU

MagneFire, 2022-11-06:
Thank you for opening this issue.

Can you also provide a journal log (`journalctl --no-pager`)?
I suspect that a library broke during the update which is not entirely uncommon unfortunately (https://github.com/AsteroidOS/asteroid/issues/226, https://github.com/AsteroidOS/asteroid/issues/225, https://github.com/AsteroidOS/asteroid/issues/224).

jackkaback, 2022-11-17:
I've since reflashed the watch so that I could continue to use it. If it happens again, I'll get you the information

eLtMosen, 2023-02-22:
Attributing this to the opkg issues we generally have when updating single packages.
https://github.com/AsteroidOS/asteroid/issues/224

closed 2023-02-22 by eLtMosen