AsteroidOS/meta-smartwatch #145: Mtp does not work in dory

Adamixdex, 2023-01-14:
it dosent work and there is no adb pull/push

Adamixdex, 2023-01-14:
forgot to add that wallpeper selection , watchface selection and home selection are not working too

MagneFire, 2023-01-15:
> it dosent work and there is no adb pull/push

Did you change the USB mode in the settings application before trying any of these? Also, are you trying these functions in combination with a Windows or a Linux machine?

> forgot to add that wallpeper selection , watchface selection and home selection are not working too

We're investigating this issue, progress is reported in: https://github.com/AsteroidOS/meta-smartwatch/issues/96

fluffysfriends, 2024-01-10:
Since #96 is marked closed and fished shouldent this issue be closed?

MagneFire, 2024-01-10:
> Since #96 is marked closed and fished shouldent this issue be closed?

That was a comment on the broken wallpaper and watchface selection. This was fixed by the Android base bump (likely an issue with Qt or libhybris).

I haven't used the MTP feature recently, but there have been some updates to the relevant packages (usb-moded, buteo-mtp) that may have fixed this issue. I'll attempt to try this soon on my `dory`.