AsteroidOS/meta-smartwatch #156: LG Watch Urbane (bass) not booting after install

aphive, 2023-02-06:
I just went through the process to install this on my LG Watch Urbane (bass), all steps passed but the watch only get to the LG logo then powers off.

Here all the steps from terminal:

```
adb reboot bootloader
```

```
fastboot oem unlock
                                                   OKAY [ 34.699s]
Finished. Total time: 34.699s
```

```
fastboot flash userdata asteroid-image-bass.ext4
Invalid sparse file format at header magic
Sending sparse 'userdata' 1/3 (261953 KB)          OKAY [ 15.632s]
Writing 'userdata'                                 OKAY [ 61.985s]
Sending sparse 'userdata' 2/3 (262140 KB)          OKAY [ 15.918s]
Writing 'userdata'                                 OKAY [ 23.076s]
Sending sparse 'userdata' 3/3 (20292 KB)           OKAY [  1.318s]
Writing 'userdata'                                 OKAY [ 51.874s]
Finished. Total time: 170.075s
```

```
fastboot flash boot zImage-dtb-bass.fastboot
Sending 'boot' (18524 KB)                          OKAY [  1.403s]
Writing 'boot'                                     OKAY [  1.703s]
Finished. Total time: 3.121s
```

```
fastboot continue
Resuming boot                                      OKAY [  0.002s]
Finished. Total time: 0.002s
```

Any help will be appreciated...

aphive, 2023-02-06:
I plugged the watch back to my computer and now it's in a boot loop...  All I get is the LG logo with a white lock icon below it.

eLtMosen, 2023-02-06:
Two things you could try. Firstly, the bass cradle is prown to bad connection leading to silent miss flashes. In that case it helps to throughly clean the contacts. Another issue might be worn pogo pins on the cradle. Pressing the watch down to the cradle during the flashing process might mitigate that. I got mine degraded to a dev device and firmly ductaped it to the cradle.
Your outputs all look fine. But multiple attempts might be necessary.
To reach the bootloader again manually, just swipe from top/left to bottom right diagonally during the first second of the boot proces. Do that not too hasty and using the thumb.

aphive, 2023-02-06:
Got it, I did a second install and I'm live..
Heart-rate monitoring and compass works fine.
I like the responsiveness of the OS, looking forward to updates....

closed 2023-02-06 by aphive