AsteroidOS/meta-smartwatch #157: Boot Loop KW88 after SPFlashTool

Esvan, 2023-02-06:
Hello, i have installed asteroid-image-harmony.ext4 with SP_Flash_Tool_v5.1924_Win.

This KW88 was running Android OS 5.1, so its not the KW88 pro. The watch is from the year 2016 to 2018 approximately.

After installing, Asteroid OS has a boot loop. In the loop it shows the logo and after 4-5 seconds it vibrates, and the logo never disappears. I can't shutdown with the button, the only way to shutdown is to connect and reinstall Asteroid OS by pressing download, after finishing the installation i unplug the cable.

Then i try to install using asteroid-image-**_inharmony_**.ext4 image and it have the same issue.

I really want to use Asteroid OS, when i was using Android 5.1 it have some beautiful skins and that's all, it has many problems but after all it's a beatiful watch, it's sound great revive my kw88 and enjoy making my own custom apps.

Thank you very much in advance.

aphive, 2023-02-06:
@Esvan How do you get to reinstall the image from boot loop?  I only get as far as it showing the boot logo then either shuts down or goes black.

Esvan, 2023-02-06:
Hí @aphive, when the watch is in the loop i plug the usb with the SPFlashTool opened, i click on download button and the KW88 stops to vibrate and the tool shows the downloading progress. After the process is finished it shutdowns when unplug, finally i press power button to turn on.

Maybe the LG is not being recogniced in the usb com port. I have my notebook with Windows 11 but i really don't know why i can't install from there, i have a older desktop computer with windows 8.1 where it recognizes that my KW88 is connected.

In my Notebook with Windows 11, the logs of SP Flash Tool shows something like this

`WARNING: EnumerateDevice(): Get friendly name failed! Err(13) (com_enum.cpp:618)`
or this
`FlashTool[12628][18344][D]: OptionDialog::InitUARTInfo(): No com port`

So try it on another computer, it worked for me.

aphive, 2023-02-06:
@Esvan  I Was able to get this working after a second install run.

For me, I got back into fastboot mode then re-flashed.

eLtMosen, 2023-02-06:
Hey @Esvan !
Sadly i am not experienced with the SPFlash Tool. I only used it once to get my KW99 up on AsteroidOS.
Also, most of the AsteroidOS developers use Linux which makes it hard to pinpoint an eventual windows issue.
The first thing i would recommend is to get a usb live Linux stick running. If you create a persistent one you can use it for all kind of tasks where Linux is needed. Since all your changes in the USB stick booted Linux will remain after reboots. (Which is not the case in regular Live Linux install images). Here is a good tutorial how to create a real Linux on a USB-stick:
Here in form of a YT video, using the Fedora distribution:
https://www.youtube.com/watch?v=-Tbob3ecg7Y

Or this one as a written step by step guide using the Linux Mint distribution (an Ubuntu derivative) which i would recommend to use since the AsteroidOS install intructions for linux use the "ubuntu commands / package manager":
https://www.fosslinux.com/60398/create-a-linux-mint-usb-drive-with-persistence.htm

Once running, first try to do the same procedure like you did in Windows using SPFlash tool and see if that makes a difference.

If you need any help with your new linux usb stick, come join our matrix channel [#Asteroid:matrix.org](https://matrix.to/#/#Asteroid:matrix.org)!

Esvan, 2023-02-07:
Hi @eLtMosen, thanks for your suggestions. 
I followed your suggestion and finally installed Fedora 37 to an external hard drive from a bootable flash drive.

I used SP_Flash_Tool_v5.1924 on Fedora and the download process on KW88, disconnected, then pressed the power button and after releasing the button the screen flashes one or two orange lines, one after the other very fast (probably the orange color is due to to the colors of the AsteroidOS icon), this time it's not a loop, and it's not vibrating either.

I tried to do a clean install, re-downloaded all the files and repeated the sequence, finally got the same result.

What did I do in Fedora 37:
At first I had a problem with a library called libpng12 to execute .sh files that I was needing to install (`sudo dnf -y install libpng12`), then I had another problem executing flash_tool.sh (SP_Flash_Tool_v5.1924) with the usb com port again which I finally fixed by running the .sh file with root permissions. Finally i installed in KW88 using the files asteroid-image-harmony.ext4, zImage-dtb-harmony.fastboot and logo.bin. 

Maybe I'm wrong on this, but can it be Fedora this time? Tomorrow I can test it with Linux Mint.

Thanks.

eLtMosen, 2023-02-07:
Ok, thank you very much for ruling out a windows issue!
Using ubuntu/mint would likely only differ in you not needing to install additional packages.
The sudo use is necessary when not having UDEV rules set up. The package to install those rules is `android-udev-rules`.

But at this point i suspect a general issue with the image and KW88 combination tbh.
Now that we are on linux, we could observe if anything usb related shows up in `dmesg` while the watch is booting.
But from your description, it might be (crashing?) too early in the boot process to have a ssh or adb connection "headless".

But, we now have at least a state where others might spark a bright idea.
@beroset is a fedora user afaiu.
And @FlorentRevest is maintaining the MTK port.

sebangc, 2023-02-14:
Hi @eLtMosen , thanks for your help. 
Today i tried dmesg on my KW88. 

This was the trace I made.
0.Install android-udev-rules  (`sudo dnf install android-udev-rules`)
2.First i reinstalled the files into the KW88, 
3.Then i cleaned the dmesg with `dmesg -c` 
4.Connect the watch and turn on
5.Wait a little and then `dmesg > log.txt`

And this was the result, its talking about error `probe of 1-1:1.0 failed with error -22`

```
[ 1776.691423] usb 1-1: new high-speed USB device number 65 using xhci_hcd
[ 1776.818787] usb 1-1: New USB device found, idVendor=0e8d, idProduct=2000, bcdDevice= 1.00
[ 1776.818802] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1776.818809] usb 1-1: Product: MT65xx Preloader
[ 1776.818814] usb 1-1: Manufacturer: MediaTek
[ 1776.843608] cdc_acm 1-1:1.0: Zero length descriptor references
[ 1776.843613] cdc_acm: probe of 1-1:1.0 failed with error -22
[ 1776.923985] cdc_acm 1-1:1.1: ttyACM0: USB ACM device
[ 1779.470933] usb 1-1: USB disconnect, device number 65
[ 1779.570824] flash_tool[6280]: segfault at fffffffffffffff8 ip 0000000000459977 sp 00007fffb7713030 error 7 in flash_tool[400000+a72000] likely on CPU 6 (core 0, socket 0)
[ 1779.570834] Code: ff 48 89 ef e8 5a 0c fc ff 48 83 c4 78 5b 5d 41 5c 41 5d 41 5e 41 5f c3 48 85 c9 48 8d 70 10 0f 84 9f 00 00 00 ba ff ff ff ff <f0> 0f c1 16 85 d2 0f 8f 5c ff ff ff 4c 89 f6 48 89 c7 48 89 4c 24

[ 1782.481433] usb 1-1: new high-speed USB device number 66 using xhci_hcd
[ 1782.609045] usb 1-1: New USB device found, idVendor=0e8d, idProduct=2000, bcdDevice= 1.00
[ 1782.609056] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1782.609061] usb 1-1: Product: MT65xx Preloader
[ 1782.609065] usb 1-1: Manufacturer: MediaTek
[ 1782.633595] cdc_acm 1-1:1.0: Zero length descriptor references
[ 1782.633608] cdc_acm: probe of 1-1:1.0 failed with error -22
[ 1782.714220] cdc_acm 1-1:1.1: ttyACM0: USB ACM device
[ 1785.261360] usb 1-1: USB disconnect, device number 66

[ 1797.889444] usb 1-1: new high-speed USB device number 67 using xhci_hcd
[ 1798.016608] usb 1-1: New USB device found, idVendor=0e8d, idProduct=2000, bcdDevice= 1.00
[ 1798.016623] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1798.016630] usb 1-1: Product: MT65xx Preloader
[ 1798.016635] usb 1-1: Manufacturer: MediaTek
[ 1798.041229] cdc_acm 1-1:1.0: Zero length descriptor references
[ 1798.041243] cdc_acm: probe of 1-1:1.0 failed with error -22
[ 1798.121624] cdc_acm 1-1:1.1: ttyACM0: USB ACM device
[ 1800.669104] usb 1-1: USB disconnect, device number 67

[ 1803.737460] usb 1-1: new high-speed USB device number 68 using xhci_hcd
```
... i tried to find something different but the data is constantly repeated in (device number...) 68, 69, 70, 71...to 81.

I was searching about this error -22, i can't figure out what does mean, on one page it talks about a missing firmware but I really don't know.

I would like to ask to @beroset about installing android-udev-rules, i don't know if this could be causing a problem for the installation of the image to KW88:

```
sudo dnf install android-udev-rules

Copr repo for PyCharm owned by phracek          5.2 kB/s | 3.6 kB     00:00    
Fedora 37 - x86_64                               99 kB/s |  51 kB     00:00    
Fedora 37 - x86_64                              9.4 MB/s |  82 MB     00:08    
Fedora Modular 37 - x86_64                      103 kB/s |  50 kB     00:00    
Fedora Modular 37 - x86_64                      1.6 MB/s | 3.8 MB     00:02    
Fedora 37 - x86_64 - Updates                     36 kB/s |  22 kB     00:00    
Fedora 37 - x86_64 - Updates                    1.0 MB/s | 5.0 MB     00:04    
Fedora Modular 37 - x86_64 - Updates             65 kB/s |  47 kB     00:00    
google-chrome                                    14 kB/s | 1.3 kB     00:00    
google-chrome                                    12 kB/s | 3.6 kB     00:00    
RPM Fusion for Fedora 37 - Nonfree - NVIDIA Dri 2.4 kB/s | 2.5 kB     00:01    
RPM Fusion for Fedora 37 - Nonfree - NVIDIA Dri  24 kB/s |  15 kB     00:00    
RPM Fusion for Fedora 37 - Nonfree - Steam      5.4 kB/s | 2.7 kB     00:00    
RPM Fusion for Fedora 37 - Nonfree - Steam      3.0 kB/s | 2.2 kB     00:00    
No match for argument: android-udev-rules
Error: Could not find any results: android-udev-rules 
```

I think it was installed well after all.

Well, that all. Thanks in advance.

Esvan, 2023-02-14:
Hi @eLtMosen , thanks for your help. 
Today i tried dmesg on my KW88. 

This was the trace I made.
0.Install android-udev-rules  (`sudo dnf install android-udev-rules`)
2.First i reinstalled the files into the KW88, 
3.Then i cleaned the dmesg with `dmesg -c` 
4.Connect the watch and turn on
5.Wait a little and then `dmesg > log.txt`

And this was the result, its talking about error `probe of 1-1:1.0 failed with error -22`

```
[ 1776.691423] usb 1-1: new high-speed USB device number 65 using xhci_hcd
[ 1776.818787] usb 1-1: New USB device found, idVendor=0e8d, idProduct=2000, bcdDevice= 1.00
[ 1776.818802] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1776.818809] usb 1-1: Product: MT65xx Preloader
[ 1776.818814] usb 1-1: Manufacturer: MediaTek
[ 1776.843608] cdc_acm 1-1:1.0: Zero length descriptor references
[ 1776.843613] cdc_acm: probe of 1-1:1.0 failed with error -22
[ 1776.923985] cdc_acm 1-1:1.1: ttyACM0: USB ACM device
[ 1779.470933] usb 1-1: USB disconnect, device number 65
[ 1779.570824] flash_tool[6280]: segfault at fffffffffffffff8 ip 0000000000459977 sp 00007fffb7713030 error 7 in flash_tool[400000+a72000] likely on CPU 6 (core 0, socket 0)
[ 1779.570834] Code: ff 48 89 ef e8 5a 0c fc ff 48 83 c4 78 5b 5d 41 5c 41 5d 41 5e 41 5f c3 48 85 c9 48 8d 70 10 0f 84 9f 00 00 00 ba ff ff ff ff <f0> 0f c1 16 85 d2 0f 8f 5c ff ff ff 4c 89 f6 48 89 c7 48 89 4c 24

[ 1782.481433] usb 1-1: new high-speed USB device number 66 using xhci_hcd
[ 1782.609045] usb 1-1: New USB device found, idVendor=0e8d, idProduct=2000, bcdDevice= 1.00
[ 1782.609056] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1782.609061] usb 1-1: Product: MT65xx Preloader
[ 1782.609065] usb 1-1: Manufacturer: MediaTek
[ 1782.633595] cdc_acm 1-1:1.0: Zero length descriptor references
[ 1782.633608] cdc_acm: probe of 1-1:1.0 failed with error -22
[ 1782.714220] cdc_acm 1-1:1.1: ttyACM0: USB ACM device
[ 1785.261360] usb 1-1: USB disconnect, device number 66

[ 1797.889444] usb 1-1: new high-speed USB device number 67 using xhci_hcd
[ 1798.016608] usb 1-1: New USB device found, idVendor=0e8d, idProduct=2000, bcdDevice= 1.00
[ 1798.016623] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1798.016630] usb 1-1: Product: MT65xx Preloader
[ 1798.016635] usb 1-1: Manufacturer: MediaTek
[ 1798.041229] cdc_acm 1-1:1.0: Zero length descriptor references
[ 1798.041243] cdc_acm: probe of 1-1:1.0 failed with error -22
[ 1798.121624] cdc_acm 1-1:1.1: ttyACM0: USB ACM device
[ 1800.669104] usb 1-1: USB disconnect, device number 67

[ 1803.737460] usb 1-1: new high-speed USB device number 68 using xhci_hcd
```
... i tried to find something different but the data is constantly repeated in (device number...) 68, 69, 70, 71...to 81.

I was searching about this error -22, i can't figure out what does mean, on one page it talks about a missing firmware but I really don't know.

I would like to ask to @beroset about installing android-udev-rules, i don't know if this could be causing a problem for the installation of the image to KW88:

```
sudo dnf install android-udev-rules

Copr repo for PyCharm owned by phracek          5.2 kB/s | 3.6 kB     00:00    
Fedora 37 - x86_64                               99 kB/s |  51 kB     00:00    
Fedora 37 - x86_64                              9.4 MB/s |  82 MB     00:08    
Fedora Modular 37 - x86_64                      103 kB/s |  50 kB     00:00    
Fedora Modular 37 - x86_64                      1.6 MB/s | 3.8 MB     00:02    
Fedora 37 - x86_64 - Updates                     36 kB/s |  22 kB     00:00    
Fedora 37 - x86_64 - Updates                    1.0 MB/s | 5.0 MB     00:04    
Fedora Modular 37 - x86_64 - Updates             65 kB/s |  47 kB     00:00    
google-chrome                                    14 kB/s | 1.3 kB     00:00    
google-chrome                                    12 kB/s | 3.6 kB     00:00    
RPM Fusion for Fedora 37 - Nonfree - NVIDIA Dri 2.4 kB/s | 2.5 kB     00:01    
RPM Fusion for Fedora 37 - Nonfree - NVIDIA Dri  24 kB/s |  15 kB     00:00    
RPM Fusion for Fedora 37 - Nonfree - Steam      5.4 kB/s | 2.7 kB     00:00    
RPM Fusion for Fedora 37 - Nonfree - Steam      3.0 kB/s | 2.2 kB     00:00    
No match for argument: android-udev-rules
Error: Could not find any results: android-udev-rules 
```

I think it was installed well after all.

Well, that all. Thanks in advance.

beroset, 2023-02-15:
The `android-udev-rules` I use are not from the distribution but from here:  https://github.com/M0Rf30/android-udev-rules

eLtMosen, 2023-02-16:
@Esvan Are you sure to own the KW88, not KW88 Pro? Since the Pro might need some different scatter file for the SPFlash tool.

Esvan, 2023-02-17:
Hi to everyone, 
I've done a number of things and I hope it can give you a clue as to what might be going on.

<details>
<summary><b>1. Trying Install Images Again From Windows 11</b></summary>
<br>

1.  Trying Install Images Again From Windows 11
    1. I used again `SP_Flash_Tool_v5.1924_Win` with Administrator permissions, this time it worked better after installing `Mediatek drivers` on the system.
    2. I installed the images on KW88 but I had worse results, it no longer presents the logo and doesn't vibrate and this confused me. 
    3. This led me to look for answers from the original image of the KW88.
</details>
<details>
<summary><b>2. Search For The Original Image Of KW88</b></summary>
<br>

2. Search For The Original Image Of KW88
   1. As a note, I searched from another search engine than Google (unfortunately Google is not giving me good search results) 
   2. After an insistent search I found a page with Russian language users, and the following images for KW88:
       - The original image of KW88 called `"KW88_CS1_B_GSM_PHT_20170614"`
       - Surprised, I found an AsteroidOS image called `"FeraLab-AsteroidOS-20190308"`
       - One more AsteroidOS image called `"FeraLab_KW88-FW-v4(11-11-2017)"`
       - I found more images but I'll try these first.
   3. **Files Location**
       1. `"KW88_CS1_B_GSM_PHT_20170614"` in https://4pda.to/forum/index.php?showtopic=768858&st=3960, thanks to the external user "hrapp". In Firmware and Kernels. Firmware instructions -> Official firmware -> Official Firmware KW88 -> Archive.
       2. `"FeraLab-AsteroidOS-20190308"` in https://4pda.to/forum/index.php?showtopic=768858&st=3960#entry82557096, thanks to the external user "feravolt"
       3. `"FeraLab_KW88-FW-v4(11-11-2017)"`  in https://4pda.to/forum/index.php?showtopic=768858&st=3960, thanks to the external user "hrapp". In Firmware and Kernels. Firmware instructions -> Custom firmware -> [KW88] Feravolt_KW88_FW-v4_11112017. 

</details>
<details>
<summary><b>3. Results Of Trying Install Each Found Firmware (3 FWs)</b></summary>
<br>

I used `SP_Flash_Tool_v5.1924_Win` From `Windows 11`

3. Results Of Trying Install Each Found Firmware (3 FWs)

    A. Image `"KW88_CS1_B_GSM_PHT_20170614"`
       - Installation: KW88 responds to installation and works.
       - Problems: Upside down screen and touch in correct position
         - If a button is up I must press the touch as if it were down.
       - SO: Android
       - Ver.: Android 5.1 
       - Build Number: KW88_CS1_B_GSM_PHT_20170614
       - About -> Model Number: KW88
       - GUI: Like original KW88
       - I descovered:
         1. The battery does not charge while the image does not work, so, it will not charge even if you have connected a charger throughout the day.
         2. The original image responded to the charger indicating 1%, I kept charging it until approximately 60% and then i kept trying other images.
         3. In my case, the image is not the original, for 2 reasons: the logo did not match the one I had and for the problem of the inverted image.

    B.  Image `"FeraLab-AsteroidOS-20190308"`
       - Installation: KW88 responds to installation and works.
       - Problems: Upside down screen and touch in correct position
       - SO: AsteroidOS 
       - Ver.: AsteroidOS 1.0
       - Build Number: -
       - Info device name: -
       - GUI: Own created by AsteroidOS
      
    C.  Image `"FeraLab_KW88-FW-v4(11-11-2017)"`
       - Installation: KW88 responds to installation and works.
       - Problems: Upside down screen and touch in correct position
       - SO: Android
       - Ver.: Android 5.1
       - Build Number: FeraLab-KW88-FW-v4
       - About -> Model Number: KW88
       - GUI: Like original KW88
       - Other relevant data obtained:
       1. RAM: 768.
[... 23 lines cut ...]
        - It gives me the idea that imei.info was wrong, so be careful, mainly because it may be wrong in the chipset information, for now I don't dare to do experiments
     2. Other web pages show that the KW88 can be similar to the KW88 PRO
        - `KW88` aliexpress [https://es.aliexpress.com/item/1005005069290247.html](https://es.aliexpress.com/item/1005005069290247.html)
        - `KW88 PRO` aliexpress [https://es.aliexpress.com/item/1005005093416473.html](https://es.aliexpress.com/item/1005005069290247.html)
</details>

<details>
<summary><b>5. What happened in Linux</b></summary>
<br>
5. What happened in Linux

   1. Thanks to @beroset, I installed `android-udev-rules` on `Fedora 37` using the link.
   2. I tried again and checked if there was any behavior change when installing the AsteroidOS image: as a result KW88 presented the same symptoms.
   3. I tried installing from a different distro expecting a different result, I installed `Linux Mint 21.1-cinnamon-64bit`, and I couldn't run `SP_Flash_Tool_v5.1916`, there was no way to install `libpng12` in this distribution
      - I tried it through apt/apt-get or manually, it got complicated. They say it's not possible because newer versions of Linux aren't supporting it, only `libpng16`. They suggest using lower versions of Linux to install `libpng12` (example `Ubuntu 16`). Strange because I was able to install from `Fedora 37` (probably because it accepted downloads from third-party libraries).
[... cut ...] ese language stuff as well. I tried installing inHarmony just in case, but it didn't work either, I'll try again anyway.

- I'm glad my KW88 is working, I wonder why the image from [asteroidos.org/install/harmony](http://asteroidos.org/install/harmony) didn't work. And I'm worried about the inverted image, on the other hand, it doesn't help that the touch is in the correct position.

I did quite a bit of work on this, hopefully I can give you an idea of ​​what may be going on, thanks in advance.