AsteroidOS/meta-smartwatch #159: [lenok] Make heartbeat sensor work

GeckoEidechse, 2023-02-22:
This is a continuation of https://github.com/AsteroidOS/meta-smartwatch/issues/55#issuecomment-1060008805

While a according to https://asteroidos.org/install/lenok/ the heartbeat sensor should work and likewise [it does work for some](https://github.com/AsteroidOS/meta-smartwatch/issues/55#issuecomment-1060008829), mine simply refuses to work.

GeckoEidechse, 2023-02-22:
And in the process of writing this issue, I followed the suggestions in https://github.com/AsteroidOS/meta-smartwatch/issues/55#issuecomment-1060008826 and managed to resolve the issue for me as follows

Flashing back Wear OS version `NWS2.170620.003` from https://forum.xda-developers.com/t/2017-09-07-nws2-170620-003-fastboot-twrp-zip-factory-images.3405311/

- [Boot](https://www.androidfilehost.com/?fid=961840155545593880)
  MD5: `3f22a71d49c689cf17954bb8279c1606`
- [Recovery](https://www.androidfilehost.com/?fid=817550096634797671)
  MD5: `244d0e137b8338e9c0c868ce39dcd2f8`
- [System](https://www.androidfilehost.com/?fid=889764386195927952)
  MD5: `625120201613967fe8d3614bbb5d137c`

by running

```
fastboot flash system NWS2.170620.003_system.img 
fastboot flash recovery NWS2.170620.003_recovery.img 
fastboot flash boot NWS2.170620.003_boot.img 
fastboot continue 
```

then after updating Play services (which took forever) I finally managed to get to heart rate sensor settings in health app, ran a measurement, and it worked.

Flashing back to Asteroid using the newest nightlies at the time, i.e. 2023-02-10 with md5 checksums

```
372200b15dfa995ef8367ab48568b64c  asteroid-image-lenok.ext4
887f8451088257eb6da3f580b03cdba5  zImage-dtb-lenok.fastboot
```

and then opening the heart rate app and it successfully measured the heart rate :D

MagneFire, 2023-02-22:
Ok, cool! Thanks for confirming!
We should probably update the website like we did for `bass` too: https://asteroidos.org/install/bass/

GeckoEidechse, 2023-02-22:
I'll make an issue on the website repo then ^^

GeckoEidechse, 2023-02-22:
Self closing this GitHub issue as my issue has been resolved and documenting this has been moved to https://github.com/AsteroidOS/asteroidos.org/issues/238

closed 2023-02-22 by GeckoEidechse