AsteroidOS/meta-smartwatch #167: [catfish] Feature request: LCD backlight

sevmonster, 2023-02-28:
Is it possible with the current state of the LCD implementation (or possible at all with how it is implemented in hardware) to enable both the LED screen and dual layer LCD screen at the same time?

My proposed use case is pretty straightforward:
1. The watch is sleeping
2. Some input is fired e.g. second button press, which does nothing in Asteroid atm, or tilt sensor to replace tilt to wake
3. Turn on the LED screen at a specific brightness, displaying a solid color (or other configurable image etc)
4. Wait some specified delay
5. Turn off the LED screen 

While niche, this would give users a more viable alternative to tilt to wake or always-on display—specifically, while the LCD in the TicWatch Pro is nice, it is not legible outside of direct light, which really reduces its comfortable usecases. This has led me to keeping tilt to wake enabled even though one of the reasons I settled on the TicWatch Pr for its battery-sipping LCD. help save battery and make the dual layer LCD more legible tilt to wake.

MagneFire, 2023-04-10:
While it's definitely possible to have such a mode it requires changes to the kernel and launcher. Implementing this in a well structured way is the challenge here.
I've attached a proof of concept to show of your idea. Note: This is a quick and dirty attempt to just check if your feature request is possible to implement.

Here are some of the changes that are definitely required:
- *kernel*: Adjust the `nanohub` driver to enable 'ambient' mode (lcd) and disable 'ambient' mode.
  - The simplest method would be to add two new `/sys/` entries one enables the LCD while the other turns it off.
- *asteroid-launcher*: Add an overlay that's any kind of color (the 'LCD backlight').
- *asteroid-launcher*: Add input handling to enable this LCD backlight mode.

Here's a demo of it in action:

https://user-images.githubusercontent.com/7857908/230911605-258f36a6-17ef-4f6f-be67-b3f626566e7e.mp4

To achieve the above all I had to do is the following:
- Use `asteroid-qmltester` to show a complete white screen using the `Rectangle` QML component.
- Disable automatically switching the LCD by removing: https://github.com/AsteroidOS/meta-smartwatch/blob/db88893be418dc0443238ac3d692adfd2872874e/meta-catfish/recipes-android/android-init/android-init/init.rc#L5

sevmonster, 2023-04-12:
It's beautiful, thank you. I certainly thought it could be possible and I'm glad it is. This would actually be a feature I think the stock watch did not support with WearOS.

> Consider using a fixed brightness level as a complete black screen causes the brightness to increase to the maximum level due to the ambient light sensor picking up the screen brightness. (Some kind of feedback loop)

This is actually why I turned off automatic brightness adjustment. Is that a firmware issue or is this managed in Asteroid? I didn't think to submit an issue for this in particular.

FanboyStudios, 2026-02-24:
> > Here's a demo of it in action:
> > 
> >  20230405_231534.web.mp4 
> 
> That feature would be perfect for the unused second button.
> 

It sure would be, but even better would be an option to have custom binds in settings for all watches with multiple buttons. :)