AsteroidOS/meta-smartwatch #175: [rover/rubyfish] Add support for Ticwatch Pro 3 LTE/GPS

eLtMosen, 2023-03-30:
Recently, some community members purchased rovers. 
This issue is ment to collect findings for a collaborated porting effort.

Initial notes:
- Similar to medaka, this has a 64bit kernel with 32bit libraries.
- Similar to medaka and catfish, this has two displays.
- Pinouts for charging and data are same as for catfish. The original rover/rubyfish usb cable only has power +/- available.
- A custom dock is needed. 
[This dock has been tested as working ](https://thangs.com/designer/yesimxev/3d-model/TicWatch%20Pro%203%20Micro%20USB%20Data%20Dock-59021)
A modified version to be used with long pogo pins instead of the original catfish charger usb boards is attached at the bottom of this issue.
- Bootloader unlocks without issues
- A full [dump that has been made using TWRP is here](https://forum.xda-developers.com/t/ticwatch-pro-3-ultra-gps-rubyfish-complete-stock-twrp-backup.4468129/)
- [Kernel sources are available here](https://github.com/mobvoi/mobvoi-ticwatch-kernel/tree/mobvoi-android-msm-rover-4.9)

Modified Ticwatch 3 dock:
[twp3-rover-rubyfish-dock-modified.zip](https://github.com/AsteroidOS/meta-smartwatch/files/11112628/twp3-rover-rubyfish-dock-modified.zip)

Pinouts on the printed dock
[image]

pegelf, 2023-06-23:
I accidentally thought, the Ticwatch Pro 3 LTE/GPS war already compatible, because I haven't found a Ticwatch Pro LTE to buy anywhere and then thought, both watches were the same.

Anyhow, now I own that watch and I can test things on it :D.

I'm not good at soldering things, so I first tried to just deisolate an old usb cable, put the wires on the contacts and hold them in place by pushing the original charger on them. Then I was able to press on the charger with one hand, while I used the other to type in the commands:
[image]

Because I don't want to do that every time when I want to boot Asteroid OS in the future (when it maybe is supported), I purchased this cable from Amazon: https://www.amazon.de/gp/product/B07DJ6T3CL/

The pogo pins are compatible, but the plastic doesn't fit and the magnets on the cable have another distance than the magnets on the watch. Also the upper magnet has the wrong polarity. I opened the plastic with a spatula and then cutted away some plastik using a wire cutter:
[image]
[image]
[image]

I cutted away too much, so it is still not that comfortable to use that cable. Maybe I can create another 3D-model of a dock for people who also bought the same cable as me.

When booting to fastboot, the device looks like this:
[image]

When I tried to boot Asteroidos I got an error message, that the dtb wasn't found. That makes sense, now that I know that the watch isn't compatible (yet?).

eLtMosen, 2023-06-23:
Thanks for testing the TicWatch E&S cable! I noticed it was similar, measured the same pin layout, but never bothered to try deeper since my cable is actually in use for a TicWatch E mooneye :D
I heard through the bushes that a port is viable but currently blocked by nobody having had time to look into deeper.
Fingers crossed that might change soonish.

pegelf, 2023-07-30:
Today I booted a "TicWatch Pro 3 Ultra LTE" using my modified E&S cable and it also shows "rover" as "PRODUCT_NAME", just like the "TicWatch Pro 3 LTE" without the "Ultra".

The hardware is also almost the same. I think the main difference is a colored background lightning of the second energy saving screen, a compass sensor and some small optical changes.

So maybe Asteroid OS will already be compatible with that watch when it will be released for the non-ultra version.

Here is also a photo taken of the ultra version:
[image]

argosphil, 2023-09-04:
I've had some success porting to what was sold to me as a "TicWatch Pro 3 Ultra GPS", codenamed `rubyfish`:

[image]

Tree is here:

https://github.com/argosphil/meta-smartwatch/tree/rubyfish

It currently requires a bit of coercion to get it to boot:
* `adb shell killall sleep`
* `adb shell chroot /sdcard systemctl start multi-user.target`

If anyone can help test, I'd be delighted. Just ping me!

eLtMosen, 2023-09-05:
Btw, the TWRP recovery images linked in the [XDA thread](https://forum.xda-developers.com/t/twrp-2021-10-02-ticwatch-pro-3-gps-lte-rubyfish-rover.4341763/) seem mixed up.
I used the rubyfish image on my rover since the rover image did not work.

argosphil, 2023-09-15:
Initial PR at #227

onacvooe, 2024-01-05:
> I've had some success (with lots of very patient help from @dodoradio, @MagneFire, and @eLtMosen) porting to what was sold to me as a "TicWatch Pro 3 Ultra GPS", codenamed `rubyfish`:
> 
> ![rubyfish](https://private-user-images.githubusercontent.com/141273723/265543693-c80010f4-2d1f-47f4-bba7-620cfd28645f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ0NDgyMTEsIm5iZiI6MTcwNDQ0NzkxMSwicGF0aCI6Ii8xNDEyNzM3MjMvMjY1NTQzNjkzLWM4MDAxMGY0LTJkMWYtNDdmNC1iYmE3LTYyMGNmZDI4NjQ1Zi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwNVQwOTQ1MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01OTc3YWIxYTQ0NGFlYWVkOTJiY2NkNWVkZjk0MGU1ZTA2NWQxMjBhNjg4YjQ1ZmMwYmExYWNhOWIwOTkwYmVhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ENakJE4vCVsaHqfiVlJx14EXa_uNai2mxMWUJIS3was)
> 
> Tree is here:
> 
> https://github.com/argosphil/meta-smartwatch/tree/rubyfish
> 
> It currently requires a bit of coercion to get it to boot:
> 
> * `adb shell killall sleep`
> * `adb shell chroot /sdcard systemctl start multi-user.target`
> 
> If anyone can help test, I'd be delighted. Just ping me!

I have 2 of those watches laying around here. And I have a modded cable that works. Sooo can you point out the steps? And what does actually work ?  Thanks!

CronosCtrl, 2024-07-08:
> I accidentally thought, the Ticwatch Pro 3 LTE/GPS was already compatible, because I haven't found a Ticwatch Pro LTE to buy anywhere and then thought, both watches were the same.
> 
> Anyhow, now I own that watch and I can test things on it :D.
> 
> I'm not good at soldering things, so I first tried to just deisolate an old usb cable, put the wires on the contacts and hold them in place by pushing the original charger on them. Then I was able to hold charger and watch charger with one hand, while I used the other to type in the commands:
> [image]
> 
> Because I don't want to do that every time when I want to boot Asteroid OS in the future (when it maybe is supported), I purchased this cable from Amazon: https://www.amazon.de/gp/product/B07DJ6T3CL/
> 
> The pogo pins are compatible, but the plastic doesn't fit and the magnets on the cable have another distance than the magnets on the watch. Also the upper magnet has the wrong polarity. I opened the plastic with a spatula and then cutted away some plastic using a wire cutter:
> [image]
> [image]
> [image]
> 
> I cutted away too much, so it is still not that comfortable to use that cable. Maybe I can create another 3D-model of a dock for people who also bought the same cable as me.
> 
> When booting to fastboot, the device looks like this:
> [image]
> 
> When I tried to boot Asteroidos I got an error message, that the dtb wasn't found. That makes sense, now that I know that the watch isn't compatible (yet?).

Hey man, so does this cable work fine when plastic is cut out? 
Is it possible to flash nethunter on the watch with it??
Thanks

pegelf, 2024-07-17:
@CronosCtrl Yes, the cable also works without the plastic. But it's more difficult to use, so only cut a cable that you don't need for other things.

It also isn't necessary to cut away as much as I did. This tutorial also describes that well:
https://web.archive.org/web/20221020171528/https://wear.lunawolf.es/en/crear-custom-cable

And yes, you can install nethunter on the watch with a modded cable:

nUrBiz, 2026-01-16:
hello everyone,

[on my last update](https://github.com/nUrBiz/android-docs/blob/main/TicWatch%20Pro%203%20Ultra%20GPS%20%5Brubyfish%5D/root-superuser-administrator.md)

i've found a cable clip-able, twisting the data one it's a lot comfortable.