AsteroidOS/meta-smartwatch #179: Cannot access SSH over USB on Mooneye

fluffysfriends, 2023-04-15:
I have a problem to access the ssh shell over usb on ubuntu 22.04.2
When i run dmesg i get the following information 

[   20.696515] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[   20.697795] vhci_hcd vhci_hcd.0: devid(131074) speed(3) speed_str(high-speed)
[   20.698379] vhci_hcd vhci_hcd.0: Device attached
[   21.051395] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[   21.231301] usb 1-1: SetAddress Request (2) to port 0
[   21.268653] usb 1-1: New USB device found, idVendor=0e8d, idProduct=0afe, bcdDevice= 4.04
[   21.269350] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   21.269885] usb 1-1: Product: mooneye
[   21.270156] usb 1-1: Manufacturer: AsteroidOS
[   21.270513] usb 1-1: SerialNumber: M6500TB1H628
[   49.294777] hv_balloon: Max. dynamic memory size: 3888 MB
[   57.479363] vhci_hcd: connection closed
[   57.479658] vhci_hcd: stop threads
[   57.480687] vhci_hcd: release socket
[   57.481033] vhci_hcd: disconnect device
[   57.481423] usb 1-1: USB disconnect, device number 2
[  164.966341] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[  164.967217] vhci_hcd vhci_hcd.0: devid(131074) speed(3) speed_str(high-speed)
[  164.968192] vhci_hcd vhci_hcd.0: Device attached
[  165.311388] usb 1-1: new high-speed USB device number 3 using vhci_hcd
[  165.461427] usb 1-1: SetAddress Request (3) to port 0
[  165.504675] usb 1-1: New USB device found, idVendor=0e8d, idProduct=0a02, bcdDevice= 4.04
[  165.506922] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  165.508563] usb 1-1: Product: mooneye
[  165.509430] usb 1-1: Manufacturer: AsteroidOS
[  165.510588] usb 1-1: SerialNumber: M6500TB1H628
[  329.861706] TCP: eth0: Driver has suspect GRO implementation, TCP performance may be compromised.
[  375.139986] vhci_hcd: connection closed
[  375.140266] vhci_hcd: stop threads
[  375.141710] vhci_hcd: release socket
[  375.142160] vhci_hcd: disconnect device
[  375.142666] usb 1-1: USB disconnect, device number 3
[  393.969558] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[  393.970438] vhci_hcd vhci_hcd.0: devid(131074) speed(3) speed_str(high-speed)
[  393.971483] vhci_hcd vhci_hcd.0: Device attached
[  394.321379] usb 1-1: new high-speed USB device number 4 using vhci_hcd
[  394.471427] usb 1-1: SetAddress Request (4) to port 0
[  394.512132] usb 1-1: New USB device found, idVendor=0e8d, idProduct=0afe, bcdDevice= 4.04
[  394.514184] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  394.517394] usb 1-1: Product: mooneye
[  394.518374] usb 1-1: Manufacturer: AsteroidOS
[  394.519571] usb 1-1: SerialNumber: M6500TB1H628
[  648.514230] vhci_hcd: connection closed
[  648.515205] vhci_hcd: stop threads
[  648.525906] vhci_hcd: release socket
[  648.528070] vhci_hcd: disconnect device
[  648.535283] usb 1-1: USB disconnect, device number 4
[  664.677272] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[  664.680386] vhci_hcd vhci_hcd.0: devid(131074) speed(3) speed_str(high-speed)
[  664.681547] vhci_hcd vhci_hcd.0: Device attached
[  665.031392] usb 1-1: new high-speed USB device number 5 using vhci_hcd
[  665.181400] usb 1-1: SetAddress Request (5) to port 0
[  665.222719] usb 1-1: New USB device found, idVendor=0e8d, idProduct=0a02, bcdDevice= 4.04
[  665.224977] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  665.226907] usb 1-1: Product: mooneye
[  665.228002] usb 1-1: Manufacturer: AsteroidOS
[  665.229770] usb 1-1: SerialNumber: M6500TB1H628

And when i am trying to connect to ssh shell i get the following.

sebastian@Laptop:~$ ssh ceres@192.168.2.15
ssh: connect to host 192.168.2.15 port 22: Connection timed out

and when i am trying to run the trouble shouting command on your website.
I get the following.  
sudo dhclient -v enp0s20f0u8
Internet Systems Consortium DHCP Client 4.4.1
Copyright 2004-2018 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Cannot find device "enp0s20f0u8"
cat: /sys/class/net/enp0s20f0u8/ifindex: No such file or directory
Error getting hardware address for "enp0s20f0u8": No such device

If you think you have received this message due to a bug rather
than a configuration issue please read the section on submitting
bugs on either our web page at www.isc.org or in the README file
before submitting a bug.  These pages explain the proper
process and the information we find helpful for debugging.

exiting.

MagneFire, 2023-04-15:
This is likely a `usb-moded` issue as ssh did work in the past and likely broke after a `usb-moded` update.

This issue along with https://github.com/AsteroidOS/meta-smartwatch/issues/180 are probably duplicates of https://github.com/AsteroidOS/meta-smartwatch/issues/57.

Getting log files from the watch would probably provide a lot more info, but those are hard to retrieve since you can't access the shell on the watch.

eLtMosen, 2023-04-15:
Hmmm, i can connect via SSH without issues.
The ` [ 57.479363] vhci_hcd: connection closed` happeing at different times suggests the connection breaks? Maybe the contatcs need cleaning or the watch needs to be placed better for the cable to safely connect?

[image]

fluffysfriends, 2023-04-15:
@eLtMosen  a question. I see in your example that you uses the user root and on your website in the example uses the user ceres wich one should i use?

eLtMosen, 2023-04-15:
That does not make a difference in this case.
I had the root shell command in the history from copying things to the watches /usr/bin/ where root privileges are necessary. Sorry for the distraction.
`ssh ceres@192.168.2.15` should work. And when needing root rights, we can always do `su root`.

MagneFire, 2023-04-15:
> @eLtMosen a question. I see in your example that you uses the user root and on your website in the example uses the user ceres wich one should i use?

User doesn't matter. This appears to be a different issue.

Looking again at your logs it appears you're using `enp0s20f0u8`. Which might not be the ethernet interface that's created by the watch.

Can you send the output of `ip a`?
And perhaps try `sudo dhclient -v eth0`?

fluffysfriends, 2023-04-15:
ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 00:15:5d:75:1b:f0 brd ff:ff:ff:ff:ff:ff
    inet 172.25.13.88/20 brd 172.25.15.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::215:5dff:fe75:1bf0/64 scope link
       valid_lft forever preferred_lft forever
I have only a network card avaibly and that one is used as bridge network to connect the VM to the network

MagneFire, 2023-04-15:
Strange, this and your `dmesg` log both show that USB is `vhci_hcd: connection closed`.
Do you have any other USB types available (USB 2.0 only for example)?

fluffysfriends, 2023-04-16:
I have tryied a diffrent port and moved on from ubuntu server to ubuntu desktop same version. I have now SSH.

closed 2023-04-16 by fluffysfriends