AsteroidOS/meta-smartwatch #191: Change button mappings

MagneFire, 2023-05-19:
Some watches have additional buttons which are sometimes mapped to volume buttons.
We might want to consider altering this to some other keycode as AsteroidOS apps are now being used by other distro's too.
Having those react to volume button changes might come of as strange.

Instead a potential button mapping could be `MenuKB` (as used on `catfish`) for watches that only have one additional button. On other watches we could potentially utilize the F keys (F1, F2). But further discussion on this is preferred.

moWerk, 2026-03-18:
Related issues requesting specific features:
https://github.com/AsteroidOS/asteroid-music/issues/25
https://github.com/AsteroidOS/asteroid-launcher/issues/197
https://github.com/AsteroidOS/asteroid-stopwatch/issues/6

beroset, 2026-11-02:
Related: https://github.com/AsteroidOS/asteroid-music/issues/32