AsteroidOS/meta-smartwatch #204: [sparrow] BT HCI device not showing - nightly possibly missing firmware?

nica-f, 2023-08-01:
On a current nightly image (2nd of August 2023) booted into in non-permanent mode there seems to be an issue with initializing the BCM4343W Bluetooth part of the WiFi/BT combo chip.

The BT HCI is high speed UART and needs to be braught up via hciattach. When trying this:
```
hciattach /dev/ttyHS0 bcm43xx
```
an interface is created but hciattach gives an error:

```
root@sparrow:~# hciattach /dev/ttyHS0 bcm43xx
bcm43xx_init
Cannot open directory '/lib/firmware': No such file or directory
Patch not found, continue anyway
Set Controller UART speed to 3000000 bit/s
Device setup complete
root@sparrow:~# hciconfig 
hci0:	Type: Primary  Bus: UART
	BD Address: AA:AA:AA:AA:AA:AA  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:738 acl:0 sco:0 events:44 errors:0
	TX bytes:1137 acl:0 sco:0 commands:44 errors:0

```

This suggests that the binary firmware package that contains the 'patch' file is not installed in the nightly built image.

Though even without the patch file and the obviously wrong BT MAC address the BT interface is working, e.g. a 'hcitool lescan' shows some LE devices in the vicinity.

eLtMosen, 2023-08-02:
Thanks for the thorough initial investigation into this issue!
Adding the logs you pasted into the matrix chat also here
[logcat-sparrow-2023-08-02.txt](https://github.com/AsteroidOS/meta-smartwatch/files/12234781/logcat-sparrow-2023-08-02.txt)
[journal-sparrow-2023-08-02.txt](https://github.com/AsteroidOS/meta-smartwatch/files/12234782/journal-sparrow-2023-08-02.txt)
Maybe @MagneFire has an idea whether this might be patch and build sytem related.

eLtMosen, 2023-08-02:
Just an addition from todays testing, beroset verified that skipjack is not impacted by the problem. I tested various other watches and found that most are ok, but wren has the same issue, as expected. And also sawfish failed to get BT and showed the same behavior when trying hciattach.

nica-f, 2023-08-03:
Adding the patchram.service file from here:

https://github.com/AsteroidOS/meta-smartwatch/blob/master/meta-sparrow/recipes-android/brcm-patchram-plus/brcm-patchram-plus/patchram.service

solves the problem. But this file should come pre-installed in the ext4 image, double checking using opkg shows that the brcm-patchram-plus package is installed and should contain the file, but it is not in the ext4 image.

closed 2023-08-10 by FlorentRevest