AsteroidOS/meta-smartwatch #212: [sturgeon] Major lag all over the UI in 20230730 build

eLtMosen, 2023-08-10:
As confirmed by All-Purpose Mat in the matrix chat, the 20230730 shows a major lag all over the UI.
App upstart takes around 10 seconds for the settings app and 20 secs for the agenda.
The UI performance is choppy and full of screen tearing.

TheSunCat, 2023-08-11:
Documenting my findings:

## Symptoms:
- main GUI is still smooth
- launching apps takes 15-20s
- all apps are extremely laggy
- bluetooth takes forever to pair, then disconnects right after
- watch heats up
- battery life is very short (maybe an hour)

Sometimes the issue seems to go away after enough rebooting. 

## Observations

- `bgrm_patchram_plus` is using basically all CPU resources:
[image]

- Killing this process (requires `su`) restores app performance (still takes about a second to open) and no longer pegs CPU at 100%

Build ID is `20230730124454`

TheSunCat, 2023-08-11:
We figured out on Matrix that the treacherous process is spawned by a systemd service:
```ini
ceres@sturgeon:~$ systemctl status patchram
× patchram.service - Load firmware into bluetooth chip
     Loaded: loaded (/lib/systemd/system/patchram.service; disabled; vendor preset: disabled)
     Active: failed (Result: signal) since Fri 2023-08-11 22:38:13 UTC; 3min 31s ago
    Process: 307 ExecStartPre=/bin/sleep 8 (code=exited, status=0/SUCCESS)
    Process: 431 ExecStartPre=/usr/sbin/rfkill unblock bluetooth (code=exited, status=0/SUCCESS)
    Process: 434 ExecStart=/usr/bin/brcm_patchram_plus --baudrate 3000000 --enable_lpm --enable_hci --no2bytes --patchram /vendor/firmware/4343A0.hcd /dev/ttyHS0 (code=killed, signal=KILL)
   Main PID: 434 (code=killed, signal=KILL)
   ```

The `Main PID` field matches the PID of the process at the top of `top`. Despite the service being disabled, something seems to be starting it.

@MagneFire suggests applying [this](https://github.com/AsteroidOS/meta-smartwatch/commit/27f5837540729fc791aa0abad6cb369648f276fb) to https://github.com/AsteroidOS/meta-smartwatch/tree/master/meta-sturgeon/recipes-android/brcm-patchram-plus

closed 2023-08-15 by beroset