AsteroidOS/meta-smartwatch #215: Support 64bit platforms with 32bit userspace

dodoradio, 2023-08-12:
Casio's later devices - kingyo (WSD-F30), medaka (WSD-F21HR) and presumably also tai (GSW-H1000) - use the 64 bit exynos 7270 with a 64 bit kernel, but perplexingly run everything else in 32 bit more. This means that the binary blobs are 32 bit, and since we have to use those through hybris we end up also having to do everything in 32 bit. The issue is that the kernel doesn't compile for 32 bit, at least not easily, so that has to stay 64 bit. 

note that hoki also uses a 64 bit SOC, but the entire system, including the kernel, is run in 32 bit mode so that device entirely sidesteps this problem.

There are two proposed solutions:
## Multilib: 
This is the openembedded feature that allows 32 bit packages on otherwise 64 bit machine

This works fine, but the issue is that this is intended for devices that are 64 bit with a few edge-case 32 bit packages. This means that the watch needs `multilib-[packagename]` packages for everything other than the kernel. The watch can't run normal armv7 packages, it needs multilib armv7 packages, which are effectively identical but incompatible. This would mean that asteroid has to compile all armv7 packages twice, and that's ridiculous. 
## Multiconf:
This allows us to set up a machine that contains multiple sub-machines, and these sub-machines can have cross-dependent recipes. The issue is that I haven't gotten this to work yet. 

This setup should allow the watch to effectively be armv7, but to have its 64 bit kernel built by a 64 bit submachine, and still have this kernel integrate with all the other packages. Docs on multiconfig are short and complex, but I've been told in openembedded irc that multiconf is the better way forward.

dodoradio, 2023-08-12:
This issue is currently the blocker for medaka port https://github.com/AsteroidOS/meta-smartwatch/pull/165

Aiden-Isik, 2025-07-20:
This will also be an issue for the port I'll be working on for lucky7/sm-r861/galaxy watch fe.