AsteroidOS/meta-smartwatch #224: [hoki] wifi support

argosphil, 2023-09-01:
TL;DR: wifi appears to work with an externally-provided driver

I tried this driver: https://github.com/Dev-msm8953/vendor_qcom_opensource_wlan_prima, which builds after providing an empty `include/linux/wakelock.h` file in the kernel tree.

WiFi appears to work with it after some modifications (disable BT, try to get it to stop doing assoc and roaming in the driver, provide mac address in `/lib/firmware`, enable with `echo 1 > /dev/wcnss_wlan`, load the module, `rfkill unblock 0`, then start `hostapd` and `wpa_supplicant`, manually configure IP and route, `ssh` works)!

Now I should figure out which of these modifications are actually necessary, and whether a stock driver would be preferable.

MagneFire, 2023-09-02:
Great work!

For what it's worth these wlan kernel driver might be more suited: https://github.com/Nhannh2791/Linux/tree/master/kernel/private/msm-fossil-cw-modules/wlan since it specifically for `hoki`. However, it's not an official source.

We might be able to use `msm-wifi-enabler` here as well since you needed to execute the same commands as listed here: https://github.com/AsteroidOS/meta-asteroid/blob/master/recipes-android/msm-wifi-enabler/msm-wifi-enabler/msm-wifi-enabler.service.

Does `connmanctl` work too (https://wiki.asteroidos.org/index.php/IP_Connection) as opposed to manually executing `rfkill`, `hostapd` and `wpa_supplicant`?

argosphil, 2023-09-02:
> Does `connmanctl` work too (https://wiki.asteroidos.org/index.php/IP_Connection) as opposed to manually executing `rfkill`, `hostapd` and `wpa_supplicant`?

Yes-ish. I just tried, and got this:
```
Error /net/connman/service/wifi_*_managed_psk: Input/output error
```
But then I tried again, and it worked. Will investigate.

gittrekt, 2024-07-04:
> Great work!
>
> For what it's worth these wlan kernel driver might be more suited: https://github.com/Nhannh2791/Linux/tree/master/kernel/private/msm-fossil-cw-modules/wlan since it specifically for [...]

Hey, new to astroidos coming from postmarketos but I just got this device so doing a bit of reading on it so please excuse my lack of understanding. I also haven't a chance to setup a build environment yet so no testing done, just code reviewing. 

Couldn't we just copy beluga for this? Basically you would just have to change the board platform in the same sense as adding an additional 
```KBUILD_OPTIONS += BOARD_PLATFORM=msm8937```

So in my theory by using [Beluga wlan module](https://github.com/AsteroidOS/meta-smartwatch/blob/master/meta-beluga%2Frecipes-kernel%2Fmodules%2Flinux-wlan-modules-beluga_p.bb#L18) as an example

```EXTRA_OEMAKE = " KERNEL_SRC="${STAGING_KERNEL_DIR}" M="${S}" BOARD_PLATFORM=msm8937"```

As we can see from the [Android.mk](https://android.googlesource.com/kernel/msm-modules/wlan/+/refs/heads/android-msm-beluga-4.9-pie-wear-mr2/Android.mk) all the following boards should be supported under pronto. `# Build/Package options for 8916, 8974, 8226, 8610, 8909, 8952, 8937, 8953 targets`

beroset, 2025-03-17:
> Hey, new to astroidos coming from postmarketos but I just got this device so doing a bit of reading on it so please excuse my lack of understanding. I also haven't a chance to setup a build [...]

I tried this just today, but unfortunately, I could not get things to compile cleanly even after changing many things.  For example:
```
CORE/HDD/src/wlan_hdd_softap_tx_rx.c:1856:17: error: 'struct net_device' has no member named 'last_rx'
 1856 |    pAdapter->dev->last_rx = jiffies;
```
It might be solvable from this direction, but would take more time and require a number of patches.  Still, thanks for the suggestion!  It may still yield some value.

pettcomputers, 2026-09-03:
Quick bit of backstory since it's relevant to why I ended up this deep in the driver: this watch sat in a drawer unused for over a year, and when I finally turned it back on it grabbed a WearOS OTA update that corrupted mid-install and left it stuck in a bootloop. Turns out there's no way to recover stock WearOS on this hardware once that happens (no factory images exist publicly, and this device has no A/B backup slot), so AsteroidOS was genuinely the only way to get the watch working again rather than a from-choice OS swap. Once I was in, I figured I might as well try to fix the stuff that's never worked on hoki instead of just accepting a bare-minimum watch. Used Claude to help me dig through the kernel source and actually get WiFi working from scratch.

Got WiFi fully working on my hoki (Skagen Falster Gen 6, model DW13S1). Wrote a complete recipe set from scratch, kernel driver builds clean against the current oe-core kernel, connects, gets a real DHCP lease, survives a cold reboot with zero manual steps. Bluetooth stays on the whole time too, no kernel oops. Worked on my specific unit — should apply to the other hoki variants too since it's the same platform, but I've only tested on my own watch.

PR is up with the actual recipes: #341

Quick summary of what it took, in case anyone else is picking this up:

The stock `/vendor/lib/modules/wlan.ko` won't load (vermagic mismatch against our own kernel build), so it has to be built from source. I used `Dev-msm8953/vendor_qcom_opensource_wlan_prima`, branch `LA.UM.10.6.2.r1-02500-89xx.0`, same as this issue already pointed at.

Two real fixes needed on top of that source, neither of which I saw written up anywhere with the actual mechanics:

1. The known Qualcomm oops in `hdd_process_bt_sco_profile()` (`CORE/HDD/src/wlan_hdd_main.c`) — it dereferences `adapter->sessionId` before `adapter` gets assigned via `hdd_get_adapter()`. Moved the lookup and null check above the first use. This is a real bug in Qualcomm's own upstream CAF tree too, not something specific to this fork.

2. One I hadn't seen mentioned anywhere: the driver's own top-level Makefile re-invokes `$(MAKE) -C $(KERNEL_SOURCE) ... CC=${CC} ... modules` without quoting `$(CC)`. Our OE build sets `CC` to a compound value with hardening flags baked in (`arm-oe-linux-gnueabi-gcc -fuse-ld=bfd -fcanon-prefix-map ...`), and the unquoted re-invocation lets make word-split it, which just breaks the build outright (`No rule to make target 'use-ld=bfd'`). Quoting `CC="$(CC)" LD="$(LD)" AR="$(AR)" OBJCOPY="$(OBJCOPY)" STRIP="$(STRIP)"` in the `all::` target fixes it clean.

Also needed an empty stub for `include/linux/wakelock.h` since it got pulled from mainline and this driver still includes it in a few places — that part matches what's already mentioned in this thread, just adding the actual recipe mechanics for it.

Firmware side: the three vendor WCNSS files (`WCNSS_qcom_cfg.ini`, `WCNSS_qcom_wlan_nv.bin`, `WCNSS_wlan_dictionary.dat`) are already sitting on `/vendor/firmware/wlan/prima/` on every hoki, just need symlinking into `/lib/firmware/` and a systemd unit to open `/dev/wcnss_wlan` and `modprobe wlan` at the right point in boot — has to run after `android-init.service` and before `connman.service`, that ordering matters or it won't come up automatically.

Tested this over a real cold reboot, not just a live insmod — comes back clean every time, same IP, 0% packet loss pinging the gateway and 1.1.1.1.

One known gap: `wlan_mac.bin` (optional MAC override firmware) isn't wired up, so it falls back to a generated MAC instead of a fixed one. Cosmetic, shows up as a harmless timeout in dmesg, doesn't affect anything.