AsteroidOS/meta-smartwatch #225: [beluga] Incorrect device codename designation on web

rtx4d, 2023-09-02:
On [this](https://asteroidos.org/watches/beluga/) page we see this table:

```
OW19W6 | OPPO Watch 41mm WiFi | beluga
OW19W8 | OPPO Watch 46mm WiFi | beluga
OW19W12 | OPPO Watch 46mm LTE | belugaxl
```

Butthese models have a different designation codename:
```
OW19W6 - beluga
OW19W8 - belugaxl
OW19W12 - orca
```

But all device variants use the same hardware base - `beluga`

argosphil, 2023-09-16:
@wannaphong is this correct?

wannaphong, 2023-09-16:
> @wannaphong is this correct?

No, it look ow19w6 and ow19w8 are beluga. (I checked both watch (real watch) are show beluga in wearos and adb. https://github.com/KHwang9883/MobileModels/blob/master/brands/oppo_global_en.md#watch-series
https://github.com/oppowearoswatch/oppo-wearos-linux-kernel-open-source/blob/f79e9a382220522e7d2ef28ca5c8616500cade1f/kernel/build/compile_kernel.sh#L86

rtx4d, 2023-09-17:
> No, it look ow19w6 and ow19w8 are beluga. I checked both watch (real watch) are show beluga in wearos and adb. [...]

I think this is just hardware/software unification, but if you compare the build props you can see that 19908 uses belugaxl everywhere except the device name

wannaphong, 2023-10-01:
> > > @wannaphong is this correct?
> > 
> > 
> > No, it look ow19w6 and ow19w8 are beluga. I checked both watch (real watch) are show beluga in wearos and adb. https://github.com/KHwang9883/MobileModels/blob/master/brands/oppo_global_en.md#watch-series https://github.com/oppowearoswatch/oppo-wearos-linux-kernel-open-source/blob/f79e9a382220522e7d2ef28ca5c8616500cade1f/kernel/build/compile_kernel.sh#L86
> 
> I think this is just hardware/software unification, but if you compare the build props you can see that 19908 uses belugaxl everywhere except the device name I think it would be more correct to use this
> 
> About 19912, also you can find "orca" in `/system/bin/google_key_install.sh`

OK. I agree. 👍

closed 2023-10-01 by MagneFire