AsteroidOS/meta-smartwatch #229: [dace] TicWatch Pro 5 support

argosphil, 2023-09-23:
I think this would make for an interesting target as it has the new Qualcomm SW5100+ chipset, 2 GB of RAM and 32 GB of flash. However, I've contacted Mobvoi and they don't want to release the kernel source code, so there's no way to get at the device identifiers necessary to create a fastboot image that won't be rejected right away.

Things that would be interesting:
* `/proc/cpuinfo`, if accessible
* `/proc/config.gz`, if accessible
* anything readable in the `/sys/devices/` subdirectory
* rooting the watch in Android would also be good, but I understand it's unlikely to happen?

Harfeur, 2023-09-23:
Here are the things asked:
- [/proc/config.gz](https://github.com/AsteroidOS/meta-smartwatch/files/12707168/config.gz)
- [/proc/cpuinfo](https://github.com/AsteroidOS/meta-smartwatch/files/12707170/cpuinfo.txt)
- [/sys/devices/](https://github.com/AsteroidOS/meta-smartwatch/files/12707182/sys_devices.tar.gz) but nothing is readable without root permission.

sschuberth, 2023-12-22:
> However, I've contacted Mobvoi and they don't want to release the kernel source code

Don't they simply have to, as the Linux kernel is GPL-2.0 licensed?

Fuseteam, 2024-09-23:
they do yeah

stone-w4tch3r, 2024-10-23:
@argosphil

I contacted mobvoi support, and it turned out that their source code had been published a long time ago!

See https://github.com/mobvoi/mobvoi-ticwatch-kernel

`uranus_kernal` branch is pro 5
`mobvoi-android-msm-rover-4.9` is pro 3 series

P.S. I'm far from being an expert on linux/android and won't be able to answer your questions. Just forwarding information from mobvoi

Fuseteam, 2024-10-23:
@stone-w4tch3r does this mean the ticwatch Pro 5, Enduro and Atlas all use the same kernel?

stone-w4tch3r, 2024-10-23:
@Fuseteam being honest, idk. But since series 3 share one kernel, I suppose this is possible for series 5 too

deadYokai, 2025-07-22:
you can root device by 
1. Patching `init_boot.img` get from OTA update ([latest at moment,](https://android.googleapis.com/packages/data/ota-api/package/2269bd24a774cae37adc720d3d97bd1b76846e2b.zip) WearOS 4.0) via Magisk
2. Flash by `fastboot flash init_boot init_boot.img`
3. Install Magisk via `adb install`
4. Launch Magisk and go by instructions

cristodcgomez, 2026-02-18:
I saw that with new Asteroid 2.0 the OS is supporting secondary screens. Will that mean it should be possible support this device with the secondary screen as well?

I'd like to start trying the porting here (have some experience in the past porting ROMs in android), but I need the connector for being able to use fastboot, where can I get one?

magic-ian, 2026-04-14:
## DEVICE SUMMARY
**Model**: TicWatch Atlas
**Codename**: dace
**Platform**: Qualcomm/QTI monaco and Wear slate companion path; likely Snapdragon W5-class family
**Kernel**: 5.15.144-android13-8-gc8ae98590581-ab151, built June 13, 2024
**Android / Wear OS**: Android 13, Wear OS 4
**Build / patch level**: TMDB.240925.002/379, vendor patch 2024-11-01
**Display**: round 466x466 AMOLED, 60 Hz, density 320
**Input / haptics**: zinitix_ts, slate-spi crown/buttons, qcom-hv-haptics
**Sensors seen on stock**: lsm6dso accel/gyro, ak0991x magnetometer, ambient light, pressure, temp, PixArt PPG, psp_hr, psp_spo2
**Partition layout**: single-slot dynamic-partition device with boot, init_boot, vendor_boot, dtbo, recovery, vbmeta, vbmeta_system, super, userdata
**Boot state**: OEM unlocking allowed

## ASTEROIDOS PORTING READINESS
**Strong points**: 

- modern Qualcomm wearable platform, 
- good stock ADB visibility, 
- display/input/sensor enumeration, 
- Bluetooth working, 
- family-matching Mobvoi kernel tree present

**Main risks**: 

- bootloader unlock not fully tested yet, 
- full backup difficult from stock,
- health/sensor stack is partly proprietary

Backup / restore of existing WearOS doesn't look practical, but perhaps there is a way to ensure it is not a one-way street
Stock ADB cannot root or read raw partitions, so a true full restore image is not easy until unlock/fastboot is validated

## KERNEL SOURCE REUSE MAP
The local tree on uranus_kernel at da697a45773b is clearly related to this family, including dace:* and PRO5-* commit history
Best board-glue starting points: monaco-idp-v1.dtsi (line 1), slate.dtsi (line 1), mobvoi_custom.dtsi (line 1), uranus-keys.dtsi (line 1)
**Display references**: monaco-idp-v1-common.dtsi (line 1), monaco-sde-display.dtsi (line 1)
**Touch references**: monaco.dtsi (line 2232), zinitix.c (line 4262)
**Crown / button bridge**: slate_events_bridge.c (line 708)
**Haptics / power**: qcom-hv-haptics.c (line 4897), pm5100.dtsi (line 347), monaco-pmic.dtsi (line 170)
**Bluetooth reference**: btpower.c (line 191)
**Caveat**: this public tree is not the exact shipping kernel source. It identifies itself as 5.4.219 in Makefile (line 1) and uses a monaco product build path in build.config (line 39), while the live watch runs 5.15.144 on an arm64 kernel. It also expects missing private/... paths, so “buildable as-is” is not proven.

## NEXT STEPS

1. Verify adb reboot bootloader and fastboot visibility non-destructively.
2. Find the exact 5.15 arm64 source or extract stock vendor_boot / DTB artifacts once low-level access is available.
3. Preserve the exact stock build identity and any official OTA package as the primary restore path.
4. Start AsteroidOS bring-up around display, touch, crown/buttons, Bluetooth, and battery first.
5. Leave HR / SpO2 for a later proprietary-integration phase unless cleaner kernel-facing drivers turn up.

## REFERENCES

Main docs: https://asteroidos.readthedocs.io
Source tree: https://github.com/AsteroidOS
Device layer: https://github.com/AsteroidOS/meta-smartwatch
Porting guide: https://asteroidos.readthedocs.io/en/latest/porting-guide/
Kernel: https://github.com/mobvoi/mobvoi-ticwatch-kernel/tree/uranus_kernel
Discussion:  https://github.com/AsteroidOS/meta-smartwatch/issues/229

cristodcgomez, 2026-08-18:
is my pleasure to inform I sucessfully unlocked my ticwatch today. I'll start my tests on porting asteroidOS

cristodcgomez, 2026-08-19:
Wish me luck, it seems the first step is running 

[image]

cristodcgomez, 2026-08-19:
After some hours... there is some progres. kernel built! 

[image]

PS: I'm addressing this with AI assisting me on the patches as I never worked with Qualcomm stuff

pietrosestito, 2026-09-10:
@cristodcgomez hi! I've been working on the TicWatch Pro 5 (dace) starting from your
meta-smartwatch fork, branch `dace` (last commit `452ca18f`, "VIA1 resuelto").
Thanks a lot: your layer and the on-screen telemetry saved me weeks.

With the changes below, AsteroidOS now boots all the way to
`multi-user.target` and stays up (tested for 5+ minutes, several boots), and I
get a root shell over USB. Everything was tested with `fastboot boot` (RAM
only). The only partition I write is the rootfs one (step 7).

Here are the steps, in order:

1. **Boot from RAM with a header v2 image.** ABL ignores a header v4 image on
   `fastboot boot`, so I added `boot-ramboot-v2.img` to `dace-boot-images`. It
   concatenates the kernel, the vendor_kernel_boot ramdisk, the init_boot
   ramdisk and the DTB into one v2 image. A single `fastboot boot` then carries
   everything and writes nothing.

2. **USB serial console to read crashes.** With `dace.console=1`, `init.sh`
   creates a CDC-ACM gadget through configfs as soon as the UDC appears. With
   `console=ttyGS0`, kernel messages come out on the PC as `/dev/ttyACM0`.
   Everything below was found this way.

3. **userdata is p62, not p82.** p82 is the Pixel Watch 2 number, inherited
   from aurora; this watch only goes up to p62. I fixed it in `machine.conf`
   and in `datapart=` in the kernel cmdline.

4. **proxy-consumer turned back on**
   (`CONFIG_REGULATOR_PROXY_CONSUMER=m`, and `proxy-consumer.ko` in
   `modules.load.dace` before `arm_smmu`). The stock OTA vendor_boot
   (TMDB.240925.002) ships `proxy-consumer.ko` and loads it at #33, right
   before `arm_smmu` (#37). The GCC `-517` looks like an ordering issue: in
   stock, gcc-monaco loads at #10. I did this to match stock; I have not
   proven it is required.

5. **Load the interconnect, and the eMMC after the SMMU.**
   - Without `qnoc-monaco`, the GCC clocks and the eMMC never come up.
     `dace.qnoc=1` loads it after the main module loop.
   - `dace.latemmc=1` holds `sdhci-msm` back until `apps-smmu` is bound.
     Otherwise the eMMC is already doing DMA when the SMMU resets
     (`ADMA Err`), and the watch drops into EDL.

6. **SMMU: implement `qcom,skip-init`** (kernel patch
   `smmu-implement-skip-init.patch`).
   - The dace DT sets `qcom,skip-init` on `apps-smmu@c600000`, but the 5.15
     GKI tree lost that option. So `arm_smmu_device_reset()` rewrites SMRs,
     S2CRs and context banks that the bootloader's streams are still using:
     4 SMRs are left programmed, and the DT declares only 1 handoff SMR.
   - The patch skips that block, like the msm-5.4 uranus kernel does.
   - Clean A/B test, same everything except `arm_smmu.ko`: without it, EDL
     after 24 s, exactly after `apps-smmu: Stage-1: 36-bit VA -> 36-bit IPA`;
     with it, alive for 355 s.
   - It is non-deterministic, so a single good boot without it proves nothing.

7. **Rootfs on a raw partition.** userdata is encrypted, so the usual
   "loop file on /sdcard" cannot work. `dace.rootpart=mmcblk0p56` in `init.sh`
   mounts a partition directly as the rootfs. I use p56 (`swap`, 3 GiB),
   written with `fastboot flash swap asteroid-image-dace.ext4`. Wear OS still
   boots fine for me, but this does overwrite that partition.

8. **The random EDL in multi-user was `coresight-tmc`.** Once systemd started,
   the watch fell into EDL at random. The cause is `coresight-tmc` (the QDSS
   ETR at `8048000.tmc`):
   - loaded by udev coldplug: EDL 4 out of 4 boots;
   - blacklisted: the watch stays alive;
   - loaded by hand from a stable emergency shell: EDL in about 10 s.

   The temporary fix is `blacklist coresight_tmc` in `/etc/modprobe.d` (a
   small recipe, `dace-modprobe-blacklist`). The root cause is still open.
   Tip: `emergency.target` never crashes, because it has no udev coldplug.

9. **dace-vendor-mount: super is p7, not p80, and there are no A/B slots.**
   I re-derived the dm-linear tables from the real LP metadata (slot 0):

   | partition   | sectors | start sector |
   |-------------|---------|--------------|
   | system      | 2710904 | 2048         |
   | vendor      | 580784  | 2712952      |
   | vendor_dlkm | 124784  | 4295792      |
   | system_dlkm | 680     | 4420576      |

   I also added an ext4 magic check, so it refuses to mount garbage. The
   script also starts the modem and rebinds `cs40l26`, which is the aurora
   haptics chip; I put both behind flags, off by default. The mount works, but
   the UI still does not start (gralloc not found).

10. **Root shell in multi-user.** Do not put `console=ttyGS0` as the LAST
    `console=` on the cmdline. When ttyGS0 is `/dev/console`, agetty's vhangup
    breaks u_serial (`WARNING u_serial.c:683 gs_close`) and the port goes
    dead. Without it, root login on ttyGS0 works. Your serial client has to
    answer the `ESC[6n` terminal-size query.

11. **Build fixes on my host.**
    - qcacld-3.0 is excluded: its `Makefile.modfinal` link step overflows
      ARG_MAX here.
    - I added `HOSTCFLAGS`/`HOSTLDFLAGS` for the native host tools.

Two tips that cost me time:

- **`systemd.journald.forward_to_console=1` hides the bug.** It slows the boot
  about 15x over a 115200 console, and the races disappear. The watch looked
  "fixed" and wasn't.
- **Panics can be read from Wear OS.** After a panic the watch reboots into
  Wear OS, and `adb bugreport` contains the LAST KMSG from pstore.
  `getprop ro.boot.bootreason` already tells you the kind of crash.

Still open:

- **UI:** gralloc is not found.
- **Charger / adb:**
  - In stock, the `usb` and `battery` power supplies are created by
    `qpnp-smblite`.
  - Our aurora-built `qpnp-smblite-main.ko` panics about 10 s after `insmod`
    (bad PC inside an IRQ), the same thing you hit on 6/9.
  - The stock `qti-qbg-main.ko` can't be loaded (`module_layout` mismatch),
    and its DT node (`qbg@4f00`) is disabled in the stock DT anyway.

cristodcgomez, 2026-09-11:
@pietrosestito Yeah that commit is just WIP stuff... what are you trying to develop from there?

Even I'm not against LLM usage, please... don't do this type of comments. Several parts of that comment are clearly LLM generated and misleading. 

Also, if you want to help with porting I'm available on matrix channel, but please provide a real comment to trust you're understanding at least what are you doing...

_PD: If you're an Hermes, Openclaw or related harness we can join forces if you send a copy of .env in the home folder of your project_

cristodcgomez, 2026-09-11:
Well guys... good news as I mentioned on matrix... we have asteroidOS for our device!

[image]

STILL WIP (touchscreen, sensors, bt, etc...) but at least I can ensure ticwatch Pro 5 will have a second life

Epiczhul, 2026-10-01:
Does the LCD screen already work?