AsteroidOS/meta-smartwatch #232: [Beluga/OW19W8] Oppo Watch Found Bugs/issues/missing software features on clean/default install

NickTheOnlyNeko, 2023-11-10:
Missing Features: WiFi access or at least an App for Accessing WiFi Settings(Not that WiFi is needed but it would be a nice "Plus" to have)

Missing Features:
**1st:** Power Saver mode, but it's being worked on in [#146](https://github.com/AsteroidOS/meta-smartwatch/issues/146).
**2nd:** Customization app/option for the extra/misc button.

Broken Features: 
 **1st:** Screen doesn't go 100% Off / 100% Dark even tho I've set the "Always on Display" setting to Off
 **2nd:** One of the watch faces has it's "Seconds" hand missing(Probably a GUI/Graphics issue). 
 **3rd:** The heart beat sensor isn't being used at all by the default app and i know that it's not working because it's somehow measuring a pulse while the smart watch being set on a desk that has 0 pulse/heart beat.

100% Working Features: 
 **1st:** The Light Sensor works.
 **2nd:** Tilt to wake works.
 **3rd:** Touch.
  **4th:** Bluetooth.

Unsure sure Features: the rest of the hardware features

all in all it just needs a tiny little bit more polishing and it would be a good alternative for the Oppo Watch because it has reached EOLS(**E**nd **O**f **L**ife **S**upport)

beroset, 2023-11-14:
With 14 different issues in one report, it makes it difficult to track.  Many of these are duplicate issues.  Also, I'm not sure what you mean in your 3rd feature request.  Maybe you can draw and paste a picture of what you have in mind?

NickTheOnlyNeko, 2023-11-15:
> With 14 different issues in one report, it makes it difficult to track. Many of these are duplicate issues. Also, I'm not sure what you mean in your 3rd feature request. Maybe you can draw and paste a picture of what you have in mind?

something like this, of course this style of calendar is from the Cinnamon DE for linux but you get the jist
[image]

NickTheOnlyNeko, 2023-11-15:
> something like this, of course this style of calendar is from the Cinnamon DE for linux but you get the jist [image]

basically use the "KISS" rule "Keep It Simple (and) Stupid"

NickTheOnlyNeko, 2023-11-15:
> With 14 different issues in one report, it makes it difficult to track. Many of these are duplicate issues. Also, I'm not sure what you mean in your 3rd feature request. Maybe you can draw and paste a picture of what you have in mind?

also the whole thing is as an external/3rd party and free "QA Service"

beroset, 2023-11-15:
Since almost all of these are duplicates of existing issues, I'd suggest that you close this and replace it with a single topic issue regarding the calendar.  Also keep in mind that the screen of a watch is much smaller than that of a phone.  If you're already concerned about the difficulties faced by "people with big hands and fingers" consider that the calendar display you show is going to be really difficult to both see (for those who don't have great vision) and to use (for those of us with big fingers).  By creating a single-issue issue instead, it's much more likely to get attention and action.  Thanks!

closed 2023-11-15 by NickTheOnlyNeko

NickTheOnlyNeko, 2023-11-17:
> Since almost all of these are duplicates of existing issues, I'd suggest that you close this and replace it with a single topic issue regarding the calendar.  Also keep in mind that the screen of a [...]

I mean for the calendar it could be "Gesture and/or Arrows" navigation for scrolling through months, i don't know what else would you want to do in an calendar app other than doing your Agenda but you have an Agenda app already so it makes not alot of sence