AsteroidOS/meta-smartwatch #242: fastboot flash userdata failing with C++ error on asus zenwatch 2 sparrow

MichaelBitard, 2024-05-18:
Hi,

I'm on linux:
- fedora 39
- fastboot version 34.0.4-android-tools
- Android Debug Bridge version 1.0.41

 I have successfully unlocked the OEM:

```
sudo fastboot oem unlock
(bootloader) The bootloader is already unlocked!
```

I verified the sha512sum of the downloaded file:
```
sha512sum ~/Downloads/asteroid-image-sparrow.ext4
aaf459fd3728c91adf0e707d30d340a1ebfdbc940840eafa99b72b2aeeb64c5548e44be02d057217feb930a13969305d7049f6084676b0c04f768322096886fa
```

When I try to flash the userdata, I get the following error:
```
sudo fastboot flash userdata ~/Downloads/asteroid-image-sparrow.ext4
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::compare: __pos (which is 917456) > this->size() (which is 0)
Aborted
```

I don't know what to do next, am I missing something?

beroset, 2024-05-20:
There was and is a problem with the Fedora 39 version of fastboot that causes this error.  See https://bugzilla.redhat.com/show_bug.cgi?id=2269950

I encountered this same problem and recompiled this version of [`fastboot`](https://matrix.to/#/!IhWLkfcitdpMucYIJB:matrix.org/$J_-i_YaVAKFZZL2nbpEsbRe1iysXw7BKr-2cBfj7Uf0?via=matrix.org&via=kde.org&via=mozilla.org) that I shared via our Matrix chat.

Since this is not an AsteroidOS issue, but rather a Fedora issue, I'm closing this.  Post a message in our Matrix chat if you'd like some help.

closed 2024-05-20 by beroset

MichaelBitard, 2024-06-03:
Thanks for pointing that out!