AsteroidOS/meta-smartwatch #245: Locked the Bootloader by accident with Oppo Watch Belgugaxl OW19W8

NickTheOnlyNeko, 2024-06-12:
i need help because i think i've fcked up, the watch doesn't even want to connect to my PC through ADB nor the fastboot command the only thing i get is when i press and hold both buttons and then it connects to my PC but it identifies as [**`Bus 001 Device 025: ID 05c6:9008 Qualcomm, Inc. Gobi Wireless Modem (QDL mode)`**] when i do `lsusb`

if someone can help me feel free to msg/dm me or even send a friend request to me on discord my @ is `@nicktheneko`

mirkokral, 2025-06-13:
just a note: QDL sounds like some download mode, maybe something like EDL or just EDL? I've heard that some chips can get unlocked with EDL. Or atleast reflashed/restored. Try some EDL interface to see if it detects.

NickTheOnlyNeko, 2025-06-15:
> just a note: QDL sounds like some download mode, maybe something like EDL or just EDL? I've heard that some chips can get unlocked with EDL. Or atleast reflashed/restored. Try some EDL interface to see if it detects.

it has EDL but it's an Encrypted EDL and i can't de-crypt it sadly

you can check up on it here
https://github.com/bkerler/edl/issues/439