AsteroidOS/meta-smartwatch #249: [beluga] Sensors do not work

Jojonintendo, 2024-07-16:
Hi, I've installed AsteroidOS on my Beluga (the small one, 41mm), and I've noticed that the following things don't work:
 - Tilt-to-wake
 - Heart Rate Monitor
 - Compass

I've SSH'd into the watch to install asteroid-health and asteroid-map, I guess GPS doesn't work either but I've not tried extensively.

As per the instructions, I've made sure the watch had the latest Wear OS version, which it had. Version is H-MR2.
I couldn't find a way to get relevant logs, not sure if the Wear OS ones or AsteroidOS ones are more useful in this case.

I'd be glad to provide any additional information if needed, thanks in advance.

MagneFire, 2024-08-09:
Does a reboot solve this issue?
Also, when you open the heart rate app do you see a green light on the back of the watch?

Can you share the following logs from your watch:
- `journalctl --no-pager`:
  - `adb shell "journalctl --no-pager" > journal.log` OR
  - `ssh ceres@192.168.2.15 "journalctl --no-pager" > journal.log`
- `/system/bin/logcat`
  - `adb shell "/system/bin/logcat" > logcat.log` OR
  - `ssh ceres@192.168.2.15 "/system/bin/logcat" > logcat.log`

In this case I suspect that the logcat would provide most information as it's usually the thing that breaks.

Jojonintendo, 2024-08-12:
A reboot doesn't seem to work, it only shuts down. Even after 5 minutes the watch is still off, I have to press the power button (bottom right one) for about 10 sec to get it to boot.

When I open the heart rate app there is no green light on the underside of the watch. Another sensor that seems to not work is ambient light sensor, as the screen always has the same brightness even in direct sunlight.

I'm uploading the logs you mentioned, hoping they can be of help. They're just after a reboot, and trying to use the watch a little, especially the heart rate app, compass, etc.
[journal.log](https://github.com/user-attachments/files/16578057/journal.log)
[logcat.log](https://github.com/user-attachments/files/16578061/logcat.log)

MagneFire, 2024-09-11:
Sorry for the delayed answer...

> A reboot doesn't seem to work, it only shuts down. Even after 5 minutes the watch is still off, I have to press the power button (bottom right one) for about 10 sec to get it to boot.
> 

Hmm, okay. I've seen this freezing issue before. After some testing it appears to be related to systemd. When using ADB it seems to work fine (`adb reboot`).

> When I open the heart rate app there is no green light on the underside of the watch. Another sensor that seems to not work is ambient light sensor, as the screen always has the same brightness even in direct sunlight.
> 

Thanks for confirming. Indeed the ambient light sensor is part of the 'sensorhub', responsible for controlling all sensors (even battery saver mode).

> I'm uploading the logs you mentioned, hoping they can be of help. They're just after a reboot, and trying to use the watch a little, especially the heart rate app, compass, etc. [...]

Thanks for providing these logs!
Unfortunately it doesn't yet reveal the actual issue, other than reporting communication issues with the 'sensorhub'.
Can you also provide the output of `getprop`? I'm wondering if it could be an issue with your model (which IIRC should work fine).

Can you also confirm if sensors work fine under WearOS?

Jojonintendo, 2024-09-11:
> Sorry for the delayed answer...

No worries at all!

> Hmm, okay. I've seen this freezing issue before. After some testing it appears to be related to systemd. When using ADB it seems to work fine (`adb reboot`).

I've just tried to "adb reboot" the watch, and it instantly reboots fine, thanks for the tip.

> Thanks for providing these logs! Unfortunately it doesn't yet reveal the actual issue, other than reporting communication issues with the 'sensorhub'. Can you also provide the output of `getprop`? I'm wondering if it could be an issue with your model (which IIRC should work fine).

I'm attaching the "getprop" output here, hoping it can shine some light on what could be missing. A quick search of "sensors" shows many entries as "false", but maybe it's just that this model doesn't have these capabilities available.

> Can you also confirm if sensors work fine under WearOS?

My beluga is now on AsteroidOS "real" install, but everything was working fine on WearOS. In fact I made sure to update it to the latest firmware too before the switch. I could try to restore it from my backup of WearOS though if needed.

[getprop.txt](https://github.com/user-attachments/files/16970298/getprop.txt)

MagneFire, 2024-09-12:
Thanks for sharing those. And thanks for helping with this effort to figure out what's causing broken sensors!

I think it might be worth it to boot into WearOS again (installing the `boot` partition should be enough).
Getting the `logcat` would be useful on WearOS to see if it also contains the communication related issues.

Another thing that WearOS does is flash the correct firmware to the MCU (which is responsible for sensors, also known a the Ambiq Micro Apollo 3 processor). There's a binary (`mcujlink`) that's responsible for flashing the correct firmware. We don't call this binary on AsteroidOS to avoid potential risks of bricking the sensors MCU (this has happened on other platforms).

Additionally, on WearOS, we can get some more useful properties via (`getprop`) that (for some reason) aren't available on AsteroidOS. Namely `ro.separate.soft` and `sys.oppo.hw_id`. These contain model information.
For reference `ro.separate.soft` can be any of the following:
- `19912`: `x19w12`
- `19908`: `x19w8`
- `19906`: `x19w6` (the expected value)
For reference `ro.separate.soft` can be any of these: `3`, `4`, `5`, `10`, `11` and maybe even more.

This would indicate that there are also at least multiple hardware revisions for the different models.

Jojonintendo, 2024-09-12:
Thanks for all the details, it's nice knowing a bit more how everything works under the hood.

I've restored WearOS on my Beluga, sensors work fine again.

I've attached both files, it seems like ro.separate.soft has the correct value. If anything else is needed, I'd be glad to help :)

[getprop.txt](https://github.com/user-attachments/files/16986104/getprop.txt)
[logcat.txt](https://github.com/user-attachments/files/16986105/logcat.txt)

Jojonintendo, 2024-09-13:
After switching to a real install of AsteroidOS again, the sensors are still working fine. Heart rate, compass, tilt and ambient light.

To me this is solved, but if you want me to do some more testing, I'd be glad to. Especially because it's not clear to me what fixed it.

Anyway, many thanks for your help and great work with AsteroidOS!

closed 2024-09-22 by MagneFire