AsteroidOS/meta-smartwatch #252: [hoki] GPS support

MagneFire, 2024-09-22:
GPS isn't currently supported on `hoki`.

Getting GPS to work probably involves the following:
- Exploring a WearOS dump to find relevant services for use with GPS/GNSS.
- Starting those services
- Observe the `/system/bin/logcat` because it will report errors which will need to be fixed.

titanism, 2025-11-06:
Any update here on GPS for hoki?

pettcomputers, 2026-09-03:
Following on from the WiFi work I posted in #224 (that issue has the full backstory on how I ended up doing this from-scratch driver digging in the first place) — GPS is fully working on my hoki now too, tested with real moving fixes outside (tracked coordinates during a drive, not just a stationary test). PR: #342

This worked on my specific hoki (Skagen Falster Gen 6, model DW13S1) — the underlying HAL/GNSS chip should be the same across all the hoki variants (Diesel Griffed, Fossil Gen 6, Michael Kors Access Gen 6, etc.) but I've only tested on my own hardware, so no promises it's identical on every model.

Same as I mentioned in #224: I used Claude Code to help me work through this — I'm not a kernel/HAL developer by trade, and this involved a lot of unfamiliar territory. Sharing it back in case it saves someone else the same digging.

The GNSS HAL service itself was actually a one-line fix, not the reverse-engineering effort this issue originally sounded like. The real vendor binary (`android.hardware.gnss@1.0-service-qti`) is already sitting on `/vendor/bin/hw/` on every hoki and runs fine standalone. The problem was that AsteroidOS's Android-HAL init compat layer (`hal-droid`) parses a `.rc` file that references the service **without** the `-qti` suffix, so it was looking for a filename that doesn't exist. Fix is a symlink:

```
ln -sf android.hardware.gnss@1.0-service-qti /vendor/bin/hw/android.hardware.gnss@1.0-service
```

Expected to also need to wire up a location-consumer stack myself — found that Beluga (same msm8937 platform family) already has this solved via the generic `geoclue` package plus `geoclue-provider-hybris-binder`, the standard libhybris bridge to the Android GNSS HAL, no device-specific hacks. Turns out hoki's tree already had this wired in (`geoclue_%.bbappend` already depends on the binder provider) — someone got that far before, so nothing to add there.

Hit one real bug getting the provider to actually start though: it's installed setuid-root by upstream's own recipe, and Linux/D-Bus strip `DBUS_SESSION_BUS_ADDRESS` from setuid processes as a real security-hardening behavior — happens below `main()`, so nothing at the invocation level could ever have worked around it. Root isn't actually needed for the GNSS binder access on this port, so I dropped the setuid bit. That fix lives in `meta-asteroid` (not this repo, not hoki-specific) — separate PR: AsteroidOS/meta-asteroid#323.

After that it was purely a testing problem, not a software one — my first two outdoor tests showed satellites being tracked (up to 15 at once) but no fix ever computed, because my own polling script was reconnecting to D-Bus on every single query, which kept tearing down and cold-restarting the GNSS session before it ever got a sustained window to actually compute a fix. Fixed the test tool to hold one connection open, and also found the actual missing piece for computing a real fix: this being a WiFi-only watch with no cellular modem, it needs the NTP-based time-injection path (not the cellular/SUPL one) to assist the fix, and that path was silently disabled because its config file didn't exist. Created `/etc/gps_xtra.ini` with `NTP_FORCE_INJECT=true` and that was it — full fix on the very next outdoor test, real coordinates updating live as I drove.

PR has the actual recipe/config changes.