AsteroidOS/meta-smartwatch #257: Hoki Fossil gen 6

Eddyw27, 2025-02-04:
Salut,
J'ai installé AstéroidOS sur ma Fossil Gen 6. L'application AsteroidoSSync est sur mon FP5 qui fonctionne sous Murena /e/os. Cependant, à chaque fois que l'app de synchronisation tente de se connecter en bluetooth à Hoki, elle plante. Du coup la connexion est impossible. Quelqu'un aurait une idée ? Ou encore mieux, une solution ?
Merci d'avance

beroset, 2025-02-07:
Voici le texte en anglais:
> I installed AsteroidOS on my Fossil Gen 6. The AsteroidOSSync app is on my FP5 which runs on Murena /e/os. However, every time the sync app tries to connect via bluetooth to Hoki, it crashes. As a result, the connection is impossible. Does anyone have an idea? Or even better, a solution?
Thanks in advance

Ce n’est pas vraiment une solution, mais est-il possible d’exécuter GadgetBridge sur votre téléphone ?

Eddyw27, 2025-02-07:
En effet, c'est ce que j'ai fini par essayé et ca permet la connection. Une fois le lien fait avec gadgetbridge, je peux connecter l'os avec asteroidossynch. Merci !