AsteroidOS/meta-smartwatch #261: [beluga] USB only working in ADB mode

beroset, 2025-02-19:
It appears that on `beluga` only ADB mode is working.

Running `systemctl status usb-moded.service` gives this result:

```
x usb-moded.service - usb-moded USB gadget controller
     Loaded: loaded (/lib/systemd/system/usb-moded.service; disabled; preset: disabled)
     Active: failed (Result: exit-code) since Fri 2024-05-31 14:33:25 EDT; 50min ago
    Process: 989 ExecStartPre=/usr/bin/init_ffs (code=exited, status=1/FAILURE)

May 31 14:33:25 beluga systemd[1]: usb-moded.service: Control process exi...LURE
May 31 14:33:25 beluga systemd[1]: usb-moded.service: Failed with result ...de'.
May 31 14:33:25 beluga systemd[1]: Failed to start usb-moded USB gadget c...ler.
May 31 14:33:25 beluga systemd[1]: usb-moded.service: Scheduled restart j...t 5.
May 31 14:33:25 beluga systemd[1]: Stopped usb-moded USB gadget controller.
May 31 14:33:25 beluga systemd[1]: usb-moded.service: Start request repea...kly.
May 31 14:33:25 beluga systemd[1]: usb-moded.service: Failed with result ...de'.
May 31 14:33:25 beluga systemd[1]: Failed to start usb-moded USB gadget c...ler.
Hint: Some lines were ellipsized, use -l to show in full.
```

Manually running `/usr/bin/init_ffs` gives this:
```
/usr/bin/init_ffs: line 3: can't create /sys/class/android_usb/android0/enable: Permission denied
/usr/bin/init_ffs: line 9: can't create /sys/class/android_usb/android0/iSerial: Permission denied
/usr/bin/init_ffs: line 10: can't create /sys/class/android_usb/android0/iManufacturer: Permission denied
/usr/bin/init_ffs: line 11: can't create /sys/class/android_usb/android0/iProduct: Permission denied
/usr/bin/init_ffs: line 13: can't create /sys/class/android_usb/android0/enable: Permission denied
```

That file is here:
```
#!/bin/sh

echo "0" > /sys/class/android_usb/android0/enable

manufacturer="$(cat /system/build.prop | grep -o 'ro.product.manufacturer=.*' | cut -d'=' -f 2)"
model="$(cat /system/build.prop | grep -o 'ro.product.model=.*' | cut -d'=' -f 2)"
serial="$(cat /proc/cmdline | sed 's/.*androidboot.serialno=//' | sed 's/ .*//')"

echo $serial       > /sys/class/android_usb/android0/iSerial
echo $manufacturer > /sys/class/android_usb/android0/iManufacturer
echo $model        > /sys/class/android_usb/android0/iProduct

echo "1" >  /sys/class/android_usb/android0/enable
```

It appears that a driver that was supposed to create `/sys/class/android_usb/android0/enable` failed to load.  This version claims to have a build id of 20231112200039, so I will build a fresh one from source and try that.

JulianDroske, 2025-03-30:
Also failed in the same result under 20250321014343.

One hint:

Newer Android kernel (like which in beluga) indeed lacks android0/enable (the whole driver, Android FunctionFS, is not a part of Linux mainline), but there's another driver called "usb_gadget" from Linux[1] (lying under /sys/kernel/config/usb_gadget, referenced by /usr/bin/init_gfs which utilizes configfs). Currently beluga's ADB seems only work on top of usb_gadget.

Working devices like sparrow[2] use init_gfs instead of init_ffs to configure usb gadget modes. However it relies on some Linux components which are currently disabled in defconfig[3].

So you may modify defconfig to enable extra usb_gadget functions and make AsteroidOS use gfs and see if helps.

[1]: https://www.kernel.org/doc/html/latest/usb/gadget_configfs.html
[2]: https://github.com/AsteroidOS/meta-smartwatch/blob/master/meta-sparrow/recipes-nemomobile/usb-moded/usb-moded/usb-moded.service#L13
[3]: https://github.com/AsteroidOS/meta-smartwatch/blob/master/meta-beluga/recipes-kernel/linux/linux-beluga/defconfig#L2925

532910, 2025-04-08:
While this is true, and ADB is the only working mode, it's impossible to change USB mode in settings and even impossible to set it to charging only