AsteroidOS/meta-smartwatch #272: [catshark] Cannot boot

rebel1725, 2025-07-21:
After the normal flashing procedure using latest firmware, my device go bootloop, the screen show a "TicWatch" and then disappear and then again the "TicWatch" logo. I tried 20250620, 20250509, 20250406, 20250307, 20250119, none worked, they all resulted in the same bootloop. TWRP boots, the stock OS works. Is there a bug?

I just unlock the bootloader, then enter fastboot, then fastboot flash userdata ~/Downloads/asteroid-image-catfish.rootfs.ext4, then fastboot flash boot ~/Downloads/zImage-dtb-catfish.fastboot, then fastboot continue, then bootloop.

Blaze10k, 2025-07-23:
Having the exact same issue, I've flashed it multiple times over and tried looking into the kmesg, dmesg and console-ramoops to see what's happening but i can't even lie i don't know what i'm looking for / at

rebel1725, 2025-07-26:
> Hold on I seem to have gotten it past the bootloop issue and I now see the asteroid logo on screen, I did this by flashing a stock vendor and aboot I found online on xda, Keep in mind I didn't do anything except try to install asteroid prior so it might just be either the watch updated while I unboxed and used it a day prior to flashing asteroid os or it's very outdated and actually needed an update in which the stock versions I flashed were up to date (I have no idea tho) Happy to provide you with a link just lmk
> 
> Forgot to add that you need to flash the stock system too for it to actually boot into the os cause only vendor and aboot will have you stuck on the logo

So could you please give me the link? Thanks a lot.

rebel1725, 2025-07-26:
Solved by flashing the stock ROM from https://mega.nz/folder/cVVBRaxL#iZbWXMs8vabDW1szJz1hgg , then following the standard flashing procedure.

closed 2025-07-26 by rebel1725