AsteroidOS/meta-smartwatch #273: [beluga] Specify exclusively `le` mode for Bluetooth

beroset, 2025-09-29:
As reported by user @dv1 on Matrix chat, the default `/etc/bluetooth/main.conf` file is empty which means that the Bluetooth controller will default to dual mode rather than `le` mode.  If this can be confirmed, it's an error.  The minimal file should be:

```
[General]
ControllerMode = le
```

MagneFire, 2025-09-29:
This is intentional, see the following for more context: https://github.com/AsteroidOS/meta-smartwatch/commit/85e529c615f87f4cd8c923116b6d37e0059b2f12

dv1, 2025-09-29:
In discussions over on Matrix, @beroset noticed that you have to use `bluetoothctl`, and inside it, use the `advertise on` command. beluga is discoverable then.